ARMmbed/mbed-os-example-lorawan #11: Lorawan feature and low power managment

atoy40, 2018-01-23:
Hello,
(This is more of a question/remark than an issue.)

I've read the code in the lorawan branch of mbed-os and I've noted Ticker are used to implement timers required by the semtech stack. Tickers, when attached, lock deepsleep. May be it can be a good idea to use LowPowerTicker, or to offer the choice between both using a configuration key (EventQueue do this).

The most "problematic" timer is the one handling dutycycle retry delay, because it can lock deepsleep for minutes when it's however a good idea to put the MCU into ! :)

thanks.
Anthony.

hasnainvirk, 2018-01-25:
@atoy40 In the next version of Mbed-OS LoRaWAN stack we will be opting to use timers from event queue itself. Currently our tests with LowPowerTimer provided with event queue are not working as expected and an investigation is due soon. However, by using timers from event queue assures us that we are just a 'flag' away from having LowPowerTimers and eventually deep sleep after the investigation is done. @0xc0170  @sg- Please hook me up with someone working with LowPowerTimer on event queue.

kivaisan, 2018-01-25:
I did some fixes to our test cases and now results look good even with EventQueue using LowPower timer.

kivaisan, 2018-01-26:
I'll close this issue. If you still have concerns about this, please reopen the issue.

closed 2018-01-26 by kivaisan