ARMmbed/mbed-os-example-lorawan #17: LoRaWAN Error given running through Windows mbed-cli

GTsapparellas, 2018-02-13:
Hi,

I'm trying to run mbed-os-example-lora on Windows using mbed-cli but I'm getting this error:

Building project mbed-os-example-lora-master (K64F, GCC_ARM)
Scan: ..
Scan: env
Scan: mbed
Scan: FEATURE_STORAGE
Scan: FEATURE_LWIP
[ERROR] Attempt to override undefined parameter 'lora.over-the-air-activation' in 'application[*]'
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\User\Downloads\mbed-os-example-lora-master\mbed-os\tools\make.py -t GCC_ARM -m K64F --source .. --build ..\BUILD\K64F\GCC_ARM" in "C:\Users\User\Downloads\mbed-os-example-lora-master\mbed-os"
---

Is this something to do with the building directory specified? Because 'lora,over-the-air-activation' parameter is actually defined inside the main_app.json file.

Thanks,

Giorgos

janjongboom, 2018-02-13:
@GTsapparellas  Which mbed-os version are you using? The error indicates that it's trying to override a parameter which is not defined in any library, which indicates that the LoRaWAN library is not present. Maybe your mbed-os points to the wrong revision?

GTsapparellas, 2018-02-13:
@janjongboom I'm using mbed-os-5.7.4 release.

mbed-os (#caeaa49d68c6, tags: latest, mbed-os-5.7.4)
  * mbed-os-5.1.0
  * mbed-os-5.1.1
  * mbed-os-5.1.2
  * mbed-os-5.1.3
  * mbed-os-5.1.4
  * mbed-os-5.1.5
  * mbed-os-5.2.0
  * mbed-os-5.2.1
  * mbed-os-5.2.2
  * mbed-os-5.2.3
  * mbed-os-5.3.0
  * mbed-os-5.3.1
  * mbed-os-5.3.2
  * mbed-os-5.3.3
  * mbed-os-5.3.4
  * mbed-os-5.3.5
  * mbed-os-5.3.6
  * mbed-os-5.4.0
  * mbed-os-5.4.1
  * mbed-os-5.4.2
  * mbed-os-5.4.3
  * mbed-os-5.4.4
  * mbed-os-5.4.5
  * mbed-os-5.4.6
  * mbed-os-5.4.7
  * mbed-os-5.5.0
  * mbed-os-5.5.1
  * mbed-os-5.5.2
  * mbed-os-5.5.3
  * mbed-os-5.5.4
  * mbed-os-5.5.5
  * mbed-os-5.5.6
  * mbed-os-5.5.7
  * mbed-os-5.6.0
  * mbed-os-5.6.1
  * mbed-os-5.6.2
  * mbed-os-5.6.3
  * mbed-os-5.6.4
  * mbed-os-5.6.5
  * mbed-os-5.6.6
  * mbed-os-5.7.0
  * mbed-os-5.7.1
  * mbed-os-5.7.2
  * mbed-os-5.7.3
  * mbed-os-5.7.4  <- current
  * mbed-os-5.7.5

Do I have to include LoRaWAN library somewhere?

janjongboom, 2018-02-13:
@GTsapparellas Yes, that's not good yet. The new LoRaWAN library will be mainlined in 5.8, so you need to switch to [the branch from this PR](https://github.com/ARMmbed/mbed-os/pull/6059) to use it.

hasnainvirk, 2018-02-13:
@GTsapparellas Please check this commit https://github.com/ARMmbed/mbed-os-example-lora/commit/93d10a42111ea391236d451d89a0eb212ea48b86

As we have not published the stack officially, application example master branch was not updated with the correct hash. Thank you for pointing us to that. Please try again with the commit above and everything should work as it should.

GTsapparellas, 2018-02-13:
@janjongboom Running got fetch hasnainvirk command I got Permission denied error.

@hasnainvirk I have imported again the mbed-os-example-lora using the latest commit as you suggested. However, at this time I'm getting the error:

Building project mbed-os-example-lora (K64F, GCC_ARM)
Scan: .
Scan: env
Scan: mbed
[ERROR] OrderedDict([(u'help', [u'Select LoRa PHY layer. See README.md for more information. Default: 0 = LORA_PHY_EU868', u'                                                                             1 = LORA_PHY_AS923', u'                                                                             2 = LORA_PHY_AU915', u'                                                                             3 = LORA_PHY_CN470', u'                                                                             4 = LORA_PHY_CN779', u'                                                                             5 = LORA_PHY_EU433', u'                                                                             6 = LORA_PHY_IN865', u'                                                                             7 = LORA_PHY_KR920', u'                                                                             8 = LORA_PHY_US915', u'                                                                             9 = LORA_PHY_US915_HYBRID']), (u'value', u'0')]) is not valid under any of the given schemas
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\User\mbed-os-example-lora\mbed-os\tools\make.py -t GCC_ARM -m K64F --source . --build .\BUILD\K64F\GCC_ARM" in "C:\Users\User\mbed-os-example-lora"
---

Trying to include the phy section inside main_app.json as you suggested on readme file does not make any difference as I'm getting another error. This time is:

C:\Users\User\mbed-os-example-lora>mbed compile -m K64F -t GCC_ARM
[ERROR] Additional properties are not allowed (u'phy' was unexpected)
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\User\mbed-os-example-lora\mbed-os\tools\make.py -t GCC_ARM -m K64F --source . --build .\BUILD\K64F\GCC_ARM" in "C:\Users\User\mbed-os-example-lora"
---

Thanks,

Giorgos

hasnainvirk, 2018-02-13:
@GTsapparellas We are aware of this issue. This was caused by a change to JSON parsing style in Mbed-OS.

* Open to features/lorawan/mbed_lib.json
* Change the 'phy' field like this:

```json 

        "phy": {
            "help": "LoRa PHY region. 0 = EU868 (default), 1 = AS923, 2 = AU915, 3 = CN470, 4 = CN779, 5 = EU433, 6 = IN865, 7 = KR920, 8 = US915, 9 = US915_HYBRID",
            "value": "0"
        },

```

hasnainvirk, 2018-02-13:
@GTsapparellas Fetching from my fork should be possible as it is public. Anyhow, if you are having issues, you can try picking up the PR directly from the Public Repository.

Assuming you are in mbed-os tree:

```sh
#6059 is the pull request ID on feature-lorawan branch
# my_branch is the new name for your local branch 
$ git fetch origin pull/6059/head:my_branch
$ git checkout my_branch
```
Now you should have a local branch 'my_branch' with our latest pull request in it on top of feature-lorawan branch. Do let us know how do you feel about the stack, architecture and API design. Happy hunting :)

hasnainvirk, 2018-02-14:
@GTsapparellas Did you try ?

GTsapparellas, 2018-02-14:
@hasnainvirk I have tried fetching your pr_branch and compile the program. This time I'm getting linkage error such as: 

Building project mbed-os-example-lora (K64F, GCC_ARM)
Scan: .
Scan: env
Scan: mbed
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Compile [  0.8%]: main.cpp
[Fatal Error] lora_radio_helper.h@18,33: netsocket/LoRaRadio.h: No such file or directory
[ERROR] In file included from .\main.cpp:25:0:
.\lora_radio_helper.h:18:33: fatal error: netsocket/LoRaRadio.h: No such file or directory
 #include "netsocket/LoRaRadio.h"
                                 ^
compilation terminated.

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\User\mbed-os-example-lora\mbed-os\tools\make.py -t GCC_ARM -m K64F --source . --build .\BUILD\K64F\GCC_ARM" in "C:\Users\User\mbed-os-example-lora"
---

LoRaRadio.h file seems not to be inside netsocket folder as it was initialised.

hasnainvirk, 2018-02-14:
@GTsapparellas We have just updated our master repository. Please clone again and deploy a clean repository. 
It's hard to tell at the moment about your baseline. So please try with the clean deployment.

GTsapparellas, 2018-02-14:
@hasnainvirk At first instance, program was given an error trying to compile LoRaWANInterface.cpp file as nwk_id was not initialised. As I am not using a network ID through The Things Network server I commented out code lines:

static uint32_t nwk_id = (MBED_CONF_LORA_DEVICE_ADDRESS & LORAWAN_NETWORK_ID_MASK);
and
connection_params.connection_u.abp.nwk_id = nwk_id;

inside lorawan_status_t LoRaWANInterface::connect() function of LoRaWANInterface.cpp file. 

Therefore, this time program compiled without any error and .bin file was copied into FRDM-K64F USB drive. On the other hand, after opening Tera Term serial monitor trying to run the program I'm getting this: 

[image]

Is this something to do with the pin configurations inside mbed_app.json file? By the way I'm using SX1272MB2xAS LoRa shield along with FRDM-K64F both consisting the LoRa Node.

Thanks,

Giogros

hasnainvirk, 2018-02-15:
@GTsapparellas Firstly, the way nwk_id is being generated is standard specific. It is not specific to TTN.
If you are using SX1272, you got to change lora-radio field like this

```
       "lora-radio": {
            "help": "Which radio to use (options: SX1272,SX1276)",
            "value": "SX1272"
        },
```

GTsapparellas, 2018-02-15:
@hasnainvirk I made your suggested change before I firstly run the program. And this is the error I was getting at first instance related to the nwk_id:

C:\Users\User\mbed-os-example-lorawan>mbed compile -m K64F -t GCC_ARM
Building project mbed-os-example-lorawan (K64F, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Compile [100.0%]: LoRaWANInterface.cpp
[Error] mbed_config.h@25,72: expected ';' before '}' token
[Error] LoRaWANInterface.cpp@83,65: expected ')' before '&' token
[ERROR] In file included from <command-line>:0:0:
.\mbed-os\features\lorawan\LoRaWANInterface.cpp: In member function 'virtual lorawan_status_t LoRaWANInterface::connect()':
./.\BUILD\K64F\GCC_ARM\mbed_config.h:25:72: error: expected ';' before '}' token
 #define MBED_CONF_LORA_DEVICE_ADDRESS                     { 0x260113D9 }                                                                                     // set by application[*]
                                                                        ^
.\mbed-os\features\lorawan\LoRaWANInterface.cpp:83:35: note: in expansion of macro 'MBED_CONF_LORA_DEVICE_ADDRESS'
         static uint32_t nwk_id = (MBED_CONF_LORA_DEVICE_ADDRESS & LORAWAN_NETWORK_ID_MASK);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.\mbed-os\features\lorawan\LoRaWANInterface.cpp:83:65: error: expected ')' before '&' token
         static uint32_t nwk_id = (MBED_CONF_LORA_DEVICE_ADDRESS & LORAWAN_NETWORK_ID_MASK);
                                                                 ^

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\User\mbed-os-example-lorawan\mbed-os\tools\make.py -t GCC_ARM -m K64F --source . --build .\BUILD\K64F\GCC_ARM" in "C:\Users\User\mbed-os-example-lorawan"
---

hasnainvirk, 2018-02-15:
@GTsapparellas How are you passing your network ID to the stack ? I mean how are you setting it up in your mbed_app.json ? 
Set your device-address which will be used to calculate your nwk_id, like:

```
"lora.device-address": "0x12345678",
```

GTsapparellas, 2018-02-15:
@hasnainvirk I have changed the way I'm passing the device address to the stack and so program is compiled without any errors this time. Thanks for that. But, when it comes to running the program again I'm getting the same error on Tera Term as above.

[image]

hasnainvirk, 2018-02-15:
@GTsapparellas Please remove `txco` pin if you are using SX1272.

hasnainvirk, 2018-02-15:
@GTsapparellas Find the fix here https://github.com/ARMmbed/mbed-os-example-lorawan/commit/b09163b52d5e6931e56da212f36f5541ccac3e12

GTsapparellas, 2018-02-15:
@hasnainvirk I've imported the b09163b revision into a clean folder and run the program again but the Tera Term error remains the same:

[image]

hasnainvirk, 2018-02-15:
This tells that you are trying to use a wrong radio driver or your pin set is not correct. What is your radio type ? Are you setting your radio type in mbed_app.json correctly ? Could you share your json with me without the keys ofcourse :)

hasnainvirk, 2018-02-15:
@GTsapparellas I went on to check your radio shield. Is it this one https://os.mbed.com/components/SX1272MB2xAS/ ?

If it is, you need to change your pin set, like below. Because I don't see any other pins connected. That's what the message on the console is saying.

```
    "config": {
        "lora-radio": {
            "help": "Which radio to use (options: SX1272,SX1276)",
            "value": "SX1276"
        },

        "lora-spi-mosi":       { "value": "D11" },
        "lora-spi-miso":       { "value": "D12" },
        "lora-spi-sclk":       { "value": "D13" },
        "lora-cs":             { "value": "D10" },
        "lora-reset":          { "value": "A0" },
        "lora-dio0":           { "value": "D2" },
        "lora-dio1":           { "value": "D3" },
        "lora-dio2":           { "value": "D4" },
        "lora-dio3":           { "value": "D5" },
        "lora-dio4":           { "value": "NC" },
        "lora-dio5":           { "value": "NC" },
        "lora-rf-switch-ctl1": { "value": "NC" },
        "lora-rf-switch-ctl2": { "value": "NC" },
        "lora-txctl":          { "value": "NC" },
        "lora-rxctl":          { "value": "NC" },
        "lora-ant-switch":     { "value": "NC" },
        "lora-pwr-amp-ctl":    { "value": "NC" },
        "lora-tcxo":           { "value": "NC" }
    },
```

GTsapparellas, 2018-02-15:
@hasnainvirk I'm using SX1272MB2xAS mbed shield along with FRDM-K64F ARM mbed board. My mbed_app.json file is set as:

[image]
 
[image]

Thanks,

Giorgos

hasnainvirk, 2018-02-15:
@GTsapparellas Okay, try this 

```
  "config": {
        "lora-radio": {
            "help": "Which radio to use (options: SX1272,SX1276)",
            "value": "SX1272"
        },

        "lora-spi-mosi":       { "value": "D11" },
        "lora-spi-miso":       { "value": "D12" },
        "lora-spi-sclk":       { "value": "D13" },
        "lora-cs":             { "value": "D10" },
        "lora-reset":          { "value": "A0" },
        "lora-dio0":           { "value": "D2" },
        "lora-dio1":           { "value": "D3" },
        "lora-dio2":           { "value": "D4" },
        "lora-dio3":           { "value": "D5" },
        "lora-dio4":           { "value": "NC" },
        "lora-dio5":           { "value": "NC" },
        "lora-rf-switch-ctl1": { "value": "NC" },
        "lora-rf-switch-ctl2": { "value": "NC" },
        "lora-txctl":          { "value": "NC" },
        "lora-rxctl":          { "value": "NC" },
        "lora-ant-switch":     { "value": "NC" },
        "lora-pwr-amp-ctl":    { "value": "NC" },
        "lora-tcxo":           { "value": "NC" }
    },
```

GTsapparellas, 2018-02-15:
@hasnainvirk I will make your suggested modifications on mbed_app.json and I will let you know in 5 minutes if that works.

GTsapparellas, 2018-02-15:
@hasnainvirk Red LED start flashing once .bin file uploaded into the board USB drive (I'm assuming this is how the program should work). But, Tera Term terminal still outputs the same error. Interestingly, when I set the baud rate to 115200 as you suggested in the documentation nothing is outputted once reset button is being pressed. Above error is happening when baud rate of the terminal is set to 9600.

hasnainvirk, 2018-02-15:
Baudrate should be 115200. Can you see your device data in your network server ?
Maybe you are not getting your console trace but the device is connected.

hasnainvirk, 2018-02-15:
@GTsapparellas By the way, if red LED is blinking, that means a hard fault.

hasnainvirk, 2018-02-15:
@GTsapparellas One more thing, you got to git reset button after flashing

GTsapparellas, 2018-02-15:
@hasnainvirk I cannot see the device data in the network server. What do you mean by hard fault? I mean I can run a simple program which continuously measures temperature value using the same boards and apparently sensor is connected to SX1272 without any problem. Do you mean that I have to hit the reset button once flashed the .bin file into the board? If yes, this is the procedure I am following until now.

hasnainvirk, 2018-02-16:
@GTsapparellas Hard fault means that your device have crashed. I am not sure what your setup is or how you are trying to use the application. I am only speculating. If you could share some of your code with me, I can try to look at it.

GTsapparellas, 2018-02-16:
@hasnainvirk The only modifications were inside the mbed_app.json as you suggested. Can you give me your e-mail in order to attach the .zip file of the project?

Thanks,

Giorgos

hasnainvirk, 2018-02-16:
@GTsapparellas hasnain.virk@arm.com

janjongboom, 2018-02-19:
So a red blinking LED indeed indicates a hard fault which is the expected outcome when a pin is wrongly used (and it might be that it switches to 9600 in that case not sure). Anyway, tested this on my K64F against latest master and it all works, no hard faults. So it looks like something weird in your project.

janjongboom, 2018-02-19:
Ah, I was testing with SX1276. I indeed see the hard fault on SX1272. Will look into it.

janjongboom, 2018-02-19:
Found the bug: https://github.com/ARMmbed/mbed-os-example-lorawan/issues/33

hasnainvirk, 2018-02-22:
@GTsapparellas As we discussed the prospects of your setup with a single channel gateway on email, I think we can close this ticket. Shall we ?

GTsapparellas, 2018-02-24:
@hasnainvirk Yes this ticket should be closed. Thanks.

closed 2018-02-26 by hasnainvirk