ARMmbed/mbed-os-example-lorawan #21: disco_l072cz_lrwan1

kleunen, 2018-02-13:
I am trying to use the example on the 'disco_l072cz_lrwan1'. I use the CLI to build the example, the online compiler does not compile the example source code. 

I first had to adjust the available memory, otherwise an error on running:
Operator new[] out of memory

I added the following to mbed_app.json:
  "thread_stack_size": {
        "value": 1024
      },
      "main_stack_size": {
        "value": 1024
      }

However now I get the following error when running the example:

Mbed LoRaWANStack initialized

 CONFIRMED message retries : 3

 Adaptive data  rate (ADR) - Enabled

 Connection - In Progress ...
Unexpected Spurious interrupt, index 2

Is it possible to run the example on the disco_l072cz_lrwan1  board, if so,how ?

hasnainvirk, 2018-02-13:
Hi @kleunen ,

Yes DISCO-L072CZ-LRWAN1 is a supported board. Seemingly, you are running out of stack here.  
Mbed LoRaWAN stack and your application, both run in 'main stack' . You are allocating too little stack for your main and OS stack sizes. Could you please try with a larger number for OS and Main stack size ?

kleunen, 2018-02-13:
Hello, 

if I increase the stack size:
"thread_stack_size": {
        "value": 2048
      },
      "main_stack_size": {
        "value": 2048
      }

 I get the following error:
Stream obj failure, errno=12

kleunen, 2018-02-13:
+----------------------------+--------+-------+-------+
| Module                     |  .text | .data |  .bss |
+----------------------------+--------+-------+-------+
| [fill]                     |    154 |     1 |    55 |
| [lib]\c.a                  |  29329 |  2472 |    89 |
| [lib]\gcc.a                |  12328 |     0 |     0 |
| [lib]\m.a                  |   1060 |     0 |     0 |
| [lib]\misc                 |    220 |    12 |    28 |
| lora_radio_helper.o        |    108 |     4 |  1760 |
| main.o                     |   1645 |     4 |   646 |
| mbed-lora-radio-drv\SX1272 |    282 |     0 |     0 |
| mbed-lora-radio-drv\SX1276 |   9511 |     0 |     1 |
| mbed-os\drivers            |   3225 |     4 |   144 |
| mbed-os\events             |   1518 |     0 |   101 |
| mbed-os\features           |  35913 |    71 |  2713 |
| mbed-os\hal                |   1629 |     4 |    66 |
| mbed-os\platform           |   3567 |     4 |   356 |
| mbed-os\rtos               |  12695 |   168 |  3941 |
| mbed-os\targets            |  12221 |     4 |  1220 |
| trace_helper.o             |    128 |     4 |   284 |
| Subtotals                  | 125533 |  2752 | 11404 |
+----------------------------+--------+-------+-------+
Total Static RAM memory (data + bss): 14156 bytes
Total Flash memory (text + data): 128285 bytes

hasnainvirk, 2018-02-13:
Could you please explain what is your setup ? 
Have you modified the application ? Because you are ending up in Stream class. I wonder why.

hasnainvirk, 2018-02-13:
@kleunen Okay, I am able to recreate that with the stack setting you are using. Let me try without these settings

janjongboom, 2018-02-14:
Default stack settings for Mbed OS are 4K now. Putting that to 2K while still retaining RTOS will probably not work.

janjongboom, 2018-02-15:
@kleunen I think the LoRa stack occupies only 6K, but with RTOS also loaded there might be very little left. It'd be interesting to see [runtime memory statistics](https://os.mbed.com/docs/v5.7/tutorials/optimizing.html#runtime-statistics) for this board.

Alt: [removing the RTOS](https://os.mbed.com/blog/entry/Reducing-memory-usage-by-tuning-RTOS-con/) on this target should save quite some memory - although not sure if the stack compiles.

kleunen, 2018-02-15:
@janjongboom The lorawan stack uses the eventqueue, so I think it can not be used without the rtos. I will have a look if I can get the runtime memory statistics, hopefully next week I have some time for this.

janjongboom, 2018-02-15:
@kleunen EventQueue works fine without RTOS. You can just run the `dispatch_forever()` call in the main loop.

Maybe need to comment out a `cmsis2.h` header somewhere but it actually compiles without it.

kleunen, 2018-02-15:
@janjongboom I think I read somewhere the rtos stack was used. Within the trace_helper I see mutex is used, but that can be quite easily removed I guess. What else is used in the lorawan stack, I can not comment on, that is something hasnainvirk should know.

janjongboom, 2018-02-15:
Here's some numbers from Multi-Tech xDot (32K RAM target). The heap stats are measured right after calling `connect()` on the stack. RTOS removal requires [one patch](https://github.com/ARMmbed/mbed-os/issues/6100) and removal of trace helpers.

**Default build (develop profile, with RTOS, default stack)**

```
ROM [|||||||||||||||||||||||                             ]   117KB/256KB
RAM [||||||||||||||||||||||||                            ]   15.4KB/32KB

Current heap: 3308 / 15652
```

**Default build without RTOS (develop profile, default stack)**

```
ROM [|||||||||||||||||||                                 ]  98.3KB/256KB
RAM [|||||||||||||                                       ]   8.82KB/32KB

Current heap: 1856 / 0
```

No clue why total heap size is not reported now... But given that it uses 7.5K less static RAM I'd expect 22K heap size.

**Release build without RTOS and with newlib-nano (default stack)**

```
ROM [|||||||||||||||                                     ]  74.3KB/256KB
RAM [|||||||||                                           ]   6.42KB/32KB

Current heap: 2308 / 0
```

Interesting that heap usage grows. Maybe newlib-nano does that.

**Release profile without RTOS and with newlib-nano and [ee-printf](https://os.mbed.com/blog/entry/Reducing-memory-usage-with-a-custom-prin/) (default stack)**

```
ROM [||||||||||||||                                      ]  72.3KB/256KB
RAM [|||||||||                                           ]   6.33KB/32KB

Current heap: 856 / 0
```

I think part of the win is that it does not need to construct any serial object here. Make sure to remove the `platform` target overrides from mbed_app.json

**Recap**

The smallest we can go to seems 7.1K of RAM usage (6.3K static, <1K dynamic). That should be plenty for the DISCO board. But it comes at a cost because the default config uses 18.7K. Maybe we should add some instructions for DISCO to use newlib-nano and no RTOS.

kleunen, 2018-02-15:
It seems only removing the rtos is already enough, correct ? And this is just removing those two includes ? Make you wonder why these includes were even there ... or am I missing something here ?

kleunen, 2018-02-15:
Also the 2x 255 bytes for tx/rx, this is the serial port buffer ? The serial port is full-duplex, so a RX and a TX buffer is required, event if you are using only a single thread. 255 bytes does sound like quite a lot actually, but ok.

janjongboom, 2018-02-15:
> It seems only removing the rtos is already enough, correct ? And this is just removing those two includes ? Make you wonder why these includes were even there ... or am I missing something here ?

Yeah, I created a bug report: https://github.com/ARMmbed/mbed-os/issues/6100

> Also the 2x 255 bytes for tx/rx, this is the serial port buffer ? The serial port is full-duplex, so a RX and a TX buffer is required, event if you are using only a single thread. 255 bytes does sound like quite a lot actually, but ok.

No, LoRaWAN buffers (see main.cpp).

hasnainvirk, 2018-02-15:
@kleunen I think Jan has explained everything that can be done. In our testing we could actually fit into the board by reducing main stack size to 2K and using ARMCC or IAR as compiler. We do not alter Mbed-OS stack size. In an alternative scenario, which Jan has described above, we can go as low as 6K but that needs some alterations. In response to 
```
What else is used in the lorawan stack, I can not comment on, that is something hasnainvirk should know.
```
The stack itself and the radio drivers are built to work without RTOS. The stack does not do any dynamic memory allocation. trace_helper() is just an application extension and you are free to remove it.

kleunen, 2018-02-20:
Indeed it compiles and runs now, but I am getting an 'OTAA Failed - Check Keys' error on activation. The keys seem to be correct, i tried different byte orders, using default phy and running on the things network. No luck so far.

kivaisan, 2018-02-20:
@kleunen fe57839bb24a455c8ba09f823d13911228c79fa9 broke few pins of murata. I just committed a fix: cc52e60f5c5dba0d46ebd00cc91dd557bfd2ebba.

Also please note that this board does not have calibrated chrystal so your cpu frequency might be slightly off and therefore timers are not exact (at least Multitech's gateways sends it's responses right in the beginning of the RX windows, so timers must be quite exact). 

If you see problems receiving responses from gateway (especially JOIN ACCEPT), you can either:
- quick hack to expand max_sys_rx_error in LoRaMac.cpp (e.g. 10 -> 100)
or
- You can use target's trim bits to tune CPU frequency for your module:
mbed-os/targets/TARGET_STM/TARGET_STM32L0/device/stm32l0xx_hal_rcc.h:
#define RCC_HSICALIBRATION_DEFAULT     ((uint32_t)0x10U)   /* Default HSI calibration trimming value */
To see how you need to tune the frequency you can use for example blinky app and oscilloscope to measure how e.g. 1sec timer is triggering. Or if you want to measure HSI frequency, enable DEBUG_MCO and measure pin PA_8. It should be exactly 30MHz.

kleunen, 2018-02-21:
I did a pull of the latest changes, and changed max_sys_rx_error in LoRaMac.cpp (e.g. 10 -> 100):

```C++
  // Init parameters which are not set in function ResetMacParameters
    _params.sys_params.max_sys_rx_error = 100;
    _params.sys_params.min_rx_symb = 6;
    _params.sys_params.retry_num = 1;
```

But still it get the same error, just takes much longer now to return 'OTAA Failed - Check Keys'.

kivaisan, 2018-02-22:
Could you please enable traces in mbed_app.json? Add these settings:
"target.features_add": ["COMMON_PAL"],
"mbed-trace.enable": true,

And are you using the default (EU868) region?

hasnainvirk, 2018-02-22:
@kleunen just in case please update your radio driver and Mbed-OS repos.   Chekout to latest master and use mbed update command.

kleunen, 2018-02-22:
Mbed LoRaWANStack initialized

 CONFIRMED message retries : 3

 Adaptive data  rate (ADR) - Enabled
[DBG ][LSTK]: Initiating OTAA
[DBG ][LSTK]: Send Join-request..
[DBG ][LMAC]: Next Channel Idx=1, DR=5

 Connection - In Progress ...
[DBG ][LMAC]: Next Channel Idx=0, DR=4
[DBG ][LMAC]: Next Channel Idx=0, DR=3
[DBG ][LMAC]: Next Transmission in 4402 ms
[DBG ][LMAC]: Next Channel Idx=0, DR=3
[DBG ][LMAC]: Next Channel Idx=0, DR=2
[DBG ][LMAC]: Next Transmission in 14047 ms
[DBG ][LMAC]: Next Channel Idx=2, DR=2
[DBG ][LMAC]: Next Channel Idx=0, DR=1
[DBG ][LMAC]: Next Transmission in 30190 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=1
[DBG ][LMAC]: Next Channel Idx=0, DR=0
[DBG ][LMAC]: Next Transmission in 74622 ms
[DBG ][LMAC]: Next Channel Idx=2, DR=0
[DBG ][LMAC]: Next Channel Idx=0, DR=5
[DBG ][LMAC]: Next Transmission in 139934 ms
[DBG ][LMAC]: Next Channel Idx=2, DR=5
[DBG ][LMAC]: Next Channel Idx=1, DR=4
[DBG ][LMAC]: Next Channel Idx=0, DR=3
[DBG ][LMAC]: Next Transmission in 4405 ms
[DBG ][LMAC]: Next Channel Idx=0, DR=3
[DBG ][LMAC]: Next Channel Idx=0, DR=2
[DBG ][LMAC]: Next Transmission in 13057 ms
[DBG ][LMAC]: Next Channel Idx=2, DR=2
[DBG ][LMAC]: Next Channel Idx=0, DR=1
[DBG ][LMAC]: Next Transmission in 30200 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=1
[DBG ][LMAC]: Next Channel Idx=0, DR=0
[DBG ][LMAC]: Next Transmission in 74629 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=0

 OTAA Failed - Check Keys

kleunen, 2018-02-22:
Perhaps now I just don't have reception, there are some gateways, but they are about 1 km away.

kivaisan, 2018-02-22:
Is there anyway you could check from gateway whether your JOIN request is visible at that end?

kleunen, 2018-02-22:
I am afraid not, i now just use the gateways from the things network, so they are not in my control

hasnainvirk, 2018-02-22:
@kleunen You can check from TTN console if your join request was received or not

kleunen, 2018-02-22:
Yes, I did check their, and it says: never seen, on the device. So I guess no join request. But this is really that it did not receive a packet at all ? Not even with incorrect network key ?

hasnainvirk, 2018-02-22:
@kleunen Yes, it has not received any of your packets. By default we are using maximum TX power in EU region. So your device power file should be alright. It may be the case that the your radio environment is heavily shadowed. Because of LoRa modulation type, Fresnel Zone becomes important. We would like to ideally have the radio balun in the horizontal plane so that your signal do not hit the ground. Try holding your device with upright antenna. And you may also want to try to increase the number of retries per datarate.      

```json
    "nb-trials": {
            "help": "Indicates how many times join can be tried, default: 12",
            "value": 12
        },
```

This is defined in mbed_lib.json and you can override that in your mbed_app.json. Please note that this number should be a multiple of six (6 datarates in EU band).

kleunen, 2018-02-26:
outdoor it works, so, it is not the code now that is incorrect

kleunen, 2018-02-26:
just one more question, I just generated a random device eui, but the there is no other source where I can get a unique device eui ?

hasnainvirk, 2018-02-26:
@kleunen well, at least just for test purpose you could generate your fake device EUI and feed it to TTN console when you are registering your device. App EUI will be generated by TTN and hence you will get a level of Uniqueness.  Device EUIs are embedded into the hardware normally, just like a MAC address. If you don't have one, that means your hardware is for testing or prototyping only. Any production ready module, will always come with a DevEUI burnt into it.

janjongboom, 2018-02-26:
Do *not* generate random EUIs. EUIs need to be globally unique, and blocks should be bought from the IEEE. If your device does not have an EUI you can either use one from TTNs range, or use a local scoped address. Do not generate 8 random bytes and use it as EUI.

closed 2018-03-15 by kleunen

janjongboom, 2018-04-12:
Just an update here for people running into this issue, I have created a minimal example here: https://github.com/janjongboom/mbed-os-example-lorawan-minimal

```
$ mbed compile -m DISCO_L072CZ_LRWAN1 -t GCC_ARM --profile=./profiles/develop.json
```

kleunen, 2018-04-12:
@janjongboom. In your example, rx_buffer can be stored on the stack:

```C++
    uint8_t rx_buffer[LORAMAC_PHY_MAXPAYLOAD];
    int16_t retcode;
    retcode = lorawan.receive(MBED_CONF_LORA_APP_PORT, rx_buffer,
                              LORAMAC_PHY_MAXPAYLOAD,
                              MSG_CONFIRMED_FLAG|MSG_UNCONFIRMED_FLAG);

```

To further reduce static memory consumption, correct ?

Also the send does not need the buffer to be alive until TX_DONE (ownership is not transferred according to lorawan documentation). And also, for TX at least use snprintf:

```C++
int8_t tx_buffer[LORAMAC_PHY_MAXPAYLOAD];
packet_len = snprintf((char*) tx_buffer, sizeof(tx_buffer), "Dummy Sensor Value is %3.1f",
                    sensor_value);
```

janjongboom, 2018-04-12:
@kleunen Yep, there's some ways to reduce this further, but I just wanted to show the newlib-nano and no RTOS tricks.