ARMmbed/mbed-os-example-lorawan #33: SX1272 driver writes to NC marked pins

janjongboom, 2018-02-19:
In SX1272_LoRaRadio.cpp this happens:

```
            _rf_switch_ctl1 = 0;
            _rf_switch_ctl2 = 0;
            _pwr_amp_ctl = 0;
            _txctl = 0;
            _rxctl = 0;
```

Which in the NXP HAL throws an assertion error when the pins are not connected. Needs to verify whether the pins are marked NC or not before doing this.

SX1276 driver does not seem to have the same issue.

This is causing the issue in #17.

hasnainvirk, 2018-02-19:
@janjongboom It's interesting to see that many STM Hal layers actually do nothing if you try to write to a NC pin. However, this NXP Hal do assert on that. In SX1276 it is handled by checking the pin being NC before performing the operation. The comment on top of the block you copied here was saying the same that the write would not do anything if the pin is not connected. Anyway, I will send a patch shortly.

hasnainvirk, 2018-02-19:
https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers/pull/11 PR made.

hasnainvirk, 2018-02-19:
Closing, as PR merged.

closed 2018-02-19 by hasnainvirk