ARMmbed/mbed-os-example-lorawan #41: Does not compile in online compiler

janjongboom, 2018-03-06:
Online compiler does not like `lorawan/bla.h` style.

janjongboom, 2018-03-06:
See for fixes https://github.com/ARMmbed/sxsw2018/commits/master

@hasnainvirk

0xc0170, 2018-03-07:
Anyone to look at this?

AnttiKauppila, 2018-03-08:
We will take care of this

kivaisan, 2018-03-08:
@janjongboom I checked your fixes, but you only had changed some of those. Do you have any guidance which qualified paths should be listed and which not. Or should we remove all qualified paths and just include the header file?

0xc0170, 2018-03-08:
> @janjongboom I checked your fixes, but you only had changed some of those. Do you have any guidance which qualified paths should be listed and which not. Or should we remove all qualified paths and just include the header file?

Those paths are used in the codebase, if you review drivers/platforms. What errors are you seeing and which files? is it just lorawan files?

janjongboom, 2018-03-08:
@0xc0170 Yes, it complains that it cannot find the header. If you remove the qualified path thing it does compile.

janjongboom, 2018-03-16:
So right now it works. Don't know if this was fixed through the release of new ARMCC, or something else, but it works.

closed 2018-03-16 by janjongboom