ARMmbed/mbed-os-example-lorawan #43: Export IAR won't work with Xdot

peksaa01, 2018-03-07:
project.py: error: MTB_MTS_XDOT not supported by iar
project.py: error: XDOT_L151CC not supported by iar

theotherjimmy, 2018-03-08:
That's not a bug

janjongboom, 2018-03-08:
@theotherjimmy Why not? It used to be impossible to build for these targets because MT only gave binaries for ARMCC and GCC ARM, but we don't use those anymore in the new LoRa stack.

theotherjimmy, 2018-03-08:
@janjongboom Missing support is not a bug. I don't see why it should be a bug. OTOH, I can't update the IAR IDE. If it does not support a target and mbed os does I would not consider that a bug.

If anything this is a feature request.

janjongboom, 2018-03-16:
@theotherjimmy But why would it fail to compile in IAR? Mbed OS can be compiled in IAR, and other ST silicon based targets build too. This just seems like somewhere in targets.json it's disabled, rather than something that doesn't work.

theotherjimmy, 2018-03-16:
@janjongboom I don't know if it would fail to compile, and I don't know if it would succeed. That's the nature of not having support.

theotherjimmy, 2018-03-16:
It looks like https://github.com/ARMmbed/mbed-os/blob/master/tools/export/iar/iar_definitions.json is missing the definitions for the STM32L151CC, which both of these targets use.

kivaisan, 2018-04-17:
This is now fixed with https://github.com/ARMmbed/mbed-os/pull/6634.

closed 2018-04-17 by kivaisan