ARMmbed/mbed-os-example-lorawan #47: Sleeping between sending data

tomhelmer, 2018-03-14:
This examples sends data as often as the duty cycle allow. Do you have an example where data is send with intervals or on sensor events ?

janjongboom, 2018-03-14:
@tomhelmer If you pause all threads Mbed OS will automatically put the device to sleep. See the [SleepManager docs](https://os.mbed.com/docs/latest/reference/sleep-manager.html).

You can change the interval in [this line](https://github.com/ARMmbed/mbed-os-example-lorawan/blob/master/main.cpp#L222). Then remove all other occurrences of `send_message`. You can also remove the interval altogether and only the send when an event happens (through an interrupt or something).

simonnilsson, 2018-04-06:
I'm also having some problems getting this to work. I have my application working as I want. It send data on both a fixed interval and on events triggered on external interrupts. 

While not sending I want to go to deepsleep and also turn of the radio completely (I have a switch controlled by a mcu pin on the board). Is it possible to somehow go to deepsleep and still have the event-queue working or does it require the processor to be in run-mode?

janjongboom, 2018-04-06:
@simonnilsson The SleepManager can tell you when it's safe to go to sleep and then you can manually toggle the right pin on the board. Hooks are listed in the [docs](https://os.mbed.com/docs/latest/reference/sleep-manager.html).

simonnilsson, 2018-04-19:
By going to stop mode I can reach about 200nA with my board (stm32l071 + sx1276). In this mode I can still wake up and toggle leds e.t.c. using external interrupts. It seems like the EventQueue stops working as I can no longer wake up on my set timers and I can not call functions from interrupts via the EventQueue. 

Shouldn't it still be able to work if using the Low-Power Timer that should continue to run in stop mode?

Another question, if you break the power to the radio, does it need to be reinitialized again and does the mbed lora stack support this?

kivaisan, 2018-04-27:
@simonnilsson Could you test if latest mbed-os master fixes your problem?

simonnilsson, 2018-05-02:
I have updated to latest mbed-os master, I have not been able to get it working if i power of the radio between transmissions, if I keep the radio on I can get the mcu down to about 40uA by letting mbed handle the sleep when the event-queue idles. I believe I can get it lower by de-initializing the internal ADC-channels i use in the code.

jmarcelino, 2018-06-05:
I'm trying to understand how to go into deep sleep between readings on the RAK811 (STM32L1 target) but failing to reach below 2mA.

Can someone post some example code how this is working/could work? I'm really struggling to understand the sleep documentation. Thanks.

janjongboom, 2018-06-06:
@jmarcelino I'm doing it like this: https://github.com/janjongboom/dsa2018-greenhouse-monitor/blob/master/1_greenhouse.cpp

But you can replace the standby call with a call to `sleep` to do lighter sleep.

jmarcelino, 2018-06-06:
@janjongboom Thanks a lot! I did have a wander around and found your example earlier, it's a great help to get new mbed users started - I now understand the mechanism and got it working on my application. 
 
Now just need some functionality to store full LoRa state to NV memory when using OTAA :-)

Thanks again.

woollettg, 2018-06-07:
@janjongboom I've tried including your standby routines in my code (which also reads a DHT device - RAK811).
The processor goes into standby ~600uA from around 6-7mA, wakes up after the specified time.  The on next standby() call however it draws 12mA and is non responsive.
I have tried sleep() and deepsleep() but they are ineffective due to being locked out by timers, SPI or something else.

It would be great if this example was more battery friendly out of the box.

jmarcelino, 2018-06-08:
@woollettg Are you sure your code is going into standby() at every possible exit? If not I think it'll might just be "hanging" there on the EventQueue waiting for a lora event.

For me it's working quite well (hopefully not a "famous last words" moment), I'm getting average 220 μA on standby() through a TPS78233 LDO

janjongboom, 2018-06-08:
@woollettg There is a sleep manager in Mbed, if you enable logging it should tell you if there are sleep locks active. Note that the device still wakes up every tick, if you enable tickless (`MBED_TICKLESS=1` in macros section in mbed_app.json) mode it uses the low-power timer and doesn't do this (if the board has all prerequisites for tickless mode).

jmarcelino, 2018-06-08:
@janjongboom TICKLESS mode doesn't compile with the  RAK811, I think it's not supported on the STM32L151?
On the other hand I don't see it waking up on every tick using your standby code, only perhaps when using  the MBED sleep() and deepsleep() functions.

janjongboom, 2018-06-08:
@jmarcelino Yes, standby mode does not require ticking. But sleep/deepsleep does.

You need to be on Mbed OS 5.9 for tickless, the STM32 tickless changes are not in 5.8.

hasnainvirk, 2018-06-20:
One issue could be the software timeouts in the radio drivers. These timeouts are not low power timers. We may need to some aliasing magic in the radio drivers which will enable low power timers being used there as well. I will update this thread once I have something testable.

hasnainvirk, 2018-06-28:
@simonnilsson @jmarcelino If you have something generic enough related to deep sleep, please contribute to this project.

tpet93, 2018-07-04:
@hasnainvirk 
Could you please elaborate briefly on what issues may occur by using LowPowerTimers for the radio drivers?
My initial testing appears to work however I may be missing something I'm not looking for. 
i.e. does  a lower timer resolution increase the risk of corrupt or missed down-links?