ARMmbed/mbed-os-example-lorawan #49: Downlink causing error respone

thiseldo, 2018-03-18:
I am running the example on both the xDot DK and the B-L072Z-LRWAN1 LoRa®Discovery kit.

I am connecting successfully to The Things Network and can upload data without issue. However when I try to send a downlink message I get the indication but the callback shows an error that indicates it has not received anything. I see the following in the output when connected to the dev boards:

```
Dummy Sensor Value = 4.3 
[INFO][LSTK]: RTS = 25 bytes, PEND = 0 
[DBG ][LMAC]: Next Channel Idx=5, DR=5 
[DBG ][LSTK]: Frame scheduled to TX..

25 bytes scheduled for transmission

Received message from Network Server nothing to read

receive() - Error code -1001 
[DBG ][LSTK]: Ack received.

Message Sent to Network Server
```

I have tried this many times and always the same issue. This is using the latest revision at this time, Revision 46fe769 in the online IDE.

Thanks

Andrew

kivaisan, 2018-03-19:
Hi Andrew, could you please check that you read from the correct port?

janjongboom, 2018-03-19:
As @kivaisan says, you need to specify the port from which you want to read, otherwise the stack returns with (I think) WOULD_BLOCK.

thiseldo, 2018-03-19:
OK, thanks. It only works if I know what port to check. Is there somewhere that holds the port number that the downlink data is sent to?

kivaisan, 2018-03-19:
@thiseldo I don't think there is that info at the moment. If you want, you can remove the port check for your application from here:
https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/LoRaWANStack.cpp#L712

janjongboom, 2018-03-19:
@kivaisan I've said this to @hasnainvirk numerous times. I think the API design for receive is broken. This is non-sensical. We should just return the port in an out parameter or something.

LoRaWAN design is such that in a downlink window there can be maximum 1 message, so why the weird API design. Just give me a package whenever there is one and I'll handle the port checking in user application.

thiseldo, 2018-03-19:
@janjongboom Thank you for identifying the issue and raising the issue on the LoRaWAN stack in mbed-os.

I have other apps based on the Multitech LoRaWAN libraries don't have this requirement and I use multiple ports to differentiate between configuration data, control data or application data.

Closing this issue. Thank you.

closed 2018-03-19 by thiseldo

janjongboom, 2018-03-19:
@thiseldo Yeah you can call `receive` multiple times with different ports to get around it for now.

SKRahulSK, 2020-09-28:
Hello,
I am using the ST B-L072Z-LRWAN1 board.
Whenever I call the receive function, I am getting the below error.
**receive() - Error code -1001**

I saw that, it means the device has nothing to read.
**LORAWAN_STATUS_WOULD_BLOCK | -1001 | Stack cannot send at the moment or have nothing to read.**

And it is always checking the port 216. How do we check all the ports?

Could you please share the example on how we can receive the data from TTN?

Thanks and Regards,
Rahul