ARMmbed/mbed-os-example-lorawan #54: NUCLEO-L073RZ not responding

Lozeonjyu, 2018-03-27:
Hi,

I am using ST NUCLEO-L073RZ and put SX1272 shield on top of it. But it is not showing anything.

Any suggestions?

Many thanks

kivaisan, 2018-03-28:
Have you done pin configuration for your set up in mbed_app.json?

Lozeonjyu, 2018-03-28:
Oh, it looks like I haven't done that.

Is there any pinout map that I can check?
I checked the sx1272 shield page on Semtech website, but there is only part of the pins included.

Thanks.

kivaisan, 2018-03-29:
I don't have this module in hand, so I cannot quickly check the pins for that, but I think @janjongboom has been using it. Jan, do you have the lora config for this set up?

I would guess that it has similar set up as we already have for K64F. Just set "lora-radio": "SX1272".

janjongboom, 2018-03-29:
@kivaisan I don't have one either, but the board has Arduino headers and pinout so your suggestion should probably work.

It would be good to sanity check the device. If you throw in some code to blink the LED does this work properly with the LoRa shield attached?

Lozeonjyu, 2018-03-30:
Thanks, guys, I checked the schematic and I am running the code smoothly now.

Just have another quick question, is there anywhere I can set the frequency channel in the project or library?

I want the device ringing in on The Things Network, but the frequency band is different according to their description. (They only have part of the channels and the step between channels are different)

janjongboom, 2018-04-05:
@Lozeonjyu I have this bug report open for this: https://github.com/ARMmbed/mbed-os/issues/6385, @kivaisan or @hasnainvirk might be able to help. In the end I just manually removed the channels that TTN does not use and only used 8-15. That works. But it's annoying.

Lozeonjyu, 2018-04-05:
@janjongboom How do I cut the channels? Is there a file in the project I can make an adjustment with, or do I need to use API from outside?

kivaisan, 2018-04-12:
Closing this issue as original issue has been resolved. Please continue TTN discussion in its own report.

closed 2018-04-12 by kivaisan