ARMmbed/mbed-os-example-lorawan #55: Auto Generate DevEUI from µC Serial

samuel-puschacher, 2018-03-29:
Hello, 
is there a way, that the DevEUI is auto generated, like in the LRWAN Stack from ST?

janjongboom, 2018-03-29:
You should not generate device EUIs, they're supposed to be globally unique. If your module has an EUI you can reuse that (or read it from flash - [xdot code here](https://github.com/ARMmbed/mbed-os/issues/6132#issuecomment-366886905)), but if not, you need to get a batch from the [IEEE](https://standards.ieee.org/develop/regauth/grpmac/index.html). If you need one for development you can create a [local address](https://serverfault.com/questions/40712/what-range-of-mac-addresses-can-i-safely-use-for-my-virtual-machines), but make sure not to deploy actual devices in that range, because uniqueness cannot be guaranteed. Often LoRaWAN network servers have a function to generate a MAC address from their own block (TTN does this for example).

samuel-puschacher, 2018-03-30:
The DISCO-L072CZ-LRWAN1 / the Type ABZ is delivered with a EUI, I just want to read this out.
Is there a code for this module?

kivaisan, 2018-04-18:
@samuel-puschacher Please see https://community.st.com/thread/41876-lorawan-demo-deveui-is-not-unique. Quickly tested with two boards and it returned different values for both.

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila