ARMmbed/mbed-os-example-lorawan #58: Stream obj failure, errno=12 on Murata module

janjongboom, 2018-04-12:
Error report through e-mail from a user who is trying to run this example on the B-L0727-LRWAN1 module. I assume building with GCC ARM 6 (but waiting for confirmation). Could anyone verify that this example still runs on the Murata module? I don't have this board here. 

@hasnainvirk @kivaisan

hasnainvirk, 2018-04-12:
@janjongboom its most probably memory exhaustion problem. Kimmo will look into it.

MaartenArits, 2018-04-12:
Hi, I reported this issue: build with GCC-ARM gcc version 7.2.1, downloaded this version yesterday for Fedora

janjongboom, 2018-04-12:
@MaartenArits Note that GCC ARM 7 is not officially supported. 

Can you try `mbed compile --profile=release`. That should build with a smaller release profile, would be interesting to see if that works.

kivaisan, 2018-04-12:
Hi. The problem with this board is the size of the memory (20kB). I have never got our example to run properly on that board when compiled with GCC. @MaartenArits if you have possibility to use ARM Compiler, it works ok.

You can also reduce the default main stack size in mbed_app.json:
`"main_stack_size":     { "value": 4096 },`
Example app needs only <2kB of stack.

MaartenArits, 2018-04-12:
Nope, no output this time. I presume the same problem but without the error message.

MaartenArits, 2018-04-12:
@kivaisan I don't have a license for the ARMCC compiler so I presume this is not possible? I will check with the change you made can come back to you. If you could link me to some documentation on how to debug this that would be helpful.

kivaisan, 2018-04-12:
Sorry, this board seems to already have override for stack size:
```
        "DISCO_L072CZ_LRWAN1": {
            "main_stack_size":      3072,
```
Btw. We already have old report about the same issue. See https://github.com/ARMmbed/mbed-os-example-lorawan/issues/21. There are also some instructions how to reduce memory usage.
And also note that uncalibrated chrystal (https://github.com/ARMmbed/mbed-os-example-lorawan/issues/21#issuecomment-366937981).

janjongboom, 2018-04-12:
@MaartenArits The ARM Compiler is used in the Online Compiler, that is one way.

Alternatively you can remove the RTOS and then build with newlib-nano for this target, that will only use 7K of RAM. Here's an example: https://github.com/janjongboom/mbed-os-example-lorawan-minimal

Build with:

```
$ mbed compile -m DISCO_L072CZ_LRWAN1 -t GCC_ARM --profile=./profiles/develop.json
```

MaartenArits, 2018-04-12:
@janjongboom You suggestion works, the project is running now.

kivaisan, 2018-04-12:
@MaartenArits can we close this issue then?

MaartenArits, 2018-04-12:
@kivaisan I a sense, yes, there is a working solution. To be honest, I wouldn't. When I bought the demo board this was the place to start for me and it did not work. There is no reference to the working implementation and to make matters worse, if you read the config you get the sense the board is supported while it is not the case. A simple solution would be to make this clear in the readme file. The right solution would be to fix the problem. I'm willing to invest the time to work on this problem but you will need to explain me how to get started.

hasnainvirk, 2018-04-12:
@MaartenArits It was mentioned in a general way here https://github.com/ARMmbed/mbed-os-example-lorawan#optional-memory-optimization

MaartenArits, 2018-04-12:
@hasnainvirk You are right to some extend it is explained but I feel my remarks are still valid due to the following reasons:
- The project compiles correctly, I would expect there to be an error on building.
- The memory usage looks fine: 15100 bytes static ram used of the 20K
- The error when programming give you no clue that this is the issue
- At no point in the memory remark is made reference to the fact this demo board is affected.

This demo is targeted to beginners. I myself am very new to the platform and even with my best attentions and full dedication I ran into this issue. I did read the comment but I didn't feel it applied to my demo board due to the reasons mentioned above.

janjongboom, 2018-04-12:
@hasnainvirk @kivaisan @sg- This is an issue that we should fix. One way would be to mark this platform as 'tiny' - I believe this would build with newlib-nano / the ARMCC equivalent by default, which would actually work.

@sg- What do you think? The RAM usage of this board with Mbed OS 5.8 and the LoRaWAN stack is 15K static + 4K dynamic, so runs out of memory very quickly.

hasnainvirk, 2018-04-12:
@janjongboom I like this idea to mark it as tiny. But then it will not work with RTOS. @sg- Can we do that ?

kivaisan, 2018-04-13:
Ok. I'll investigate more if we could get example to work also with GCC.

janjongboom, 2018-04-13:
Tracking in https://github.com/ARMmbed/mbed-os/issues/6631

kivaisan, 2018-04-13:
@MaartenArits Could you test PR #59 . I got GCC compilation working with those changes, but we're really on the edge. 
**Note that due to only 1k main thread stack size, mbed traces cannot be enabled**` ("mbed-trace.enable": true)`

kivaisan, 2018-04-17:
I merged https://github.com/ARMmbed/mbed-os-example-lorawan/pull/59. I'll close this issue. Please reopen if you still have problems with this.

closed 2018-04-17 by kivaisan