ARMmbed/mbed-os-example-lorawan #75: SX1272 board not working on Nucleo-F103 (both MBed enabled boards)

Electrobub, 2018-05-26:
Hello,

I am trying to get this example code working with the SX1272MB2xAS shield attached to a Nucleo-F103 board

I have followed the issue found here: https://github.com/ARMmbed/mbed-os-example-lorawan/issues/21#issuecomment-367636001 for tips

I am trying to connect to the TTN network, have added the EUIs requested and I am receiving the following output (I have enabled traces):

```
Mbed LoRaWANStack initialized 

 CONFIRMED message retries : 3 

 Adaptive data  rate (ADR) - Enabled 
[DBG ][LSTK]: Initiating OTAA
[DBG ][LSTK]: Send Join-request..
[DBG ][LMAC]: Next Channel Idx=1, DR=5

 Connection - In Progress ...

 OTAA Failed - Check Keys 
```
In the issue thread I linked, there is output between the 

> Connection - In progress
and 

>  OTAA Failed - Check Keys 

Shouldn' t there be output between these 2 to show attempts at connections? (Trace)
(Even if I am out of range of a gateway? The GW is about 800m away from the node though)

kivaisan, 2018-05-28:
If you use the latest mbed-os, you should this kind of traces:

```
[DBG ][LSTK]: Initializing MAC layer

 Mbed LoRaWANStack initialized 

 CONFIRMED message retries : 3 

 Adaptive data  rate (ADR) - Enabled 
[DBG ][LSTK]: Initiating OTAA
[DBG ][LSTK]: Sending Join Request ...
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: TX: Channel=2, DR=5

 Connection - In Progress ...
[DBG ][LSTK]: Transmission completed
[DBG ][LMAC]: Opening RX1 Window
[DBG ][LMAC]: Opening RX2 Window
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: TX: Channel=0, DR=4
[DBG ][LSTK]: Transmission completed
[DBG ][LMAC]: Opening RX1 Window
[DBG ][LMAC]: Opening RX2 Window
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: DC enforced: Transmitting in 5012 ms
```

It looks like your radio configuration is not correct. Please check your pins are correct and you have set `SX1272 `as a `"lora-radio"`

Electrobub, 2018-05-28:
In config:
I have set: 
```
"lora-radio": {
            "help": "Which radio to use (options: SX1272,SX1276)",
            "value": "SX1272"
        },
```
And with the SX1272 shield attached, I have these pins set:
```
        "lora-spi-mosi":       { "value": "PA_7" },
        "lora-spi-miso":       { "value": "PA_6" },
        "lora-spi-sclk":       { "value": "PA_5" },
        "lora-cs":             { "value": "PB_6" },
        "lora-reset":          { "value": "PA_0" },
        "lora-dio0":           { "value": "PA_10" },
        "lora-dio1":           { "value": "PB_3" },
        "lora-dio2":           { "value": "PB_5" },
        "lora-dio3":           { "value": "PB_4" },
...
```

Which looks correct?

kivaisan, 2018-05-29:
It looks correct. I don't have F103 board, but I tested SX1272 shield with F401 with these settings (Nucleo PinMaps.h also includes mappings for Arduino pin names) and everything works ok:

```
        "NUCLEO_F401RE": {
            "lora-radio":          "SX1272",
            "lora-spi-mosi":       "D11",
            "lora-spi-miso":       "D12",
            "lora-spi-sclk":       "D13",
            "lora-cs":             "D10",
            "lora-reset":          "A0",
            "lora-dio0":           "D2",
            "lora-dio1":           "D3",
            "lora-dio2":           "D4",
            "lora-dio3":           "D5",
            "lora-dio4":           "D8",
            "lora-dio5":           "D9",
            "lora-rf-switch-ctl1": "NC",
            "lora-rf-switch-ctl2": "NC",
            "lora-txctl":          "NC",
            "lora-rxctl":          "NC",
            "lora-ant-switch":     "A4",
            "lora-pwr-amp-ctl":    "NC",
            "lora-tcxo":           "NC"
        },
```

with mbed-os 5.8.5 release you should see this kind of traces:

```
<111 ms>[DBG ][LSTK]: Initializing MAC layer

 Mbed LoRaWANStack initialized 

 CONFIRMED message retries : 3 

 Adaptive data  rate (ADR) - Enabled 
<159 ms>[DBG ][LSTK]: Initiating OTAA
<163 ms>[DBG ][LSTK]: Send Join-request..
<202 ms>[DBG ][LMAC]: Next Channel Idx=0, DR=5

 Connection - In Progress ...
<5346 ms>[DBG ][LSTK]: Join OK!

 Connection - Successful 

```

Electrobub, 2018-05-29:
I checked and changed the online compiler to use the new mbed release and now it seems to work! (It doesn't get stuck as before)
Thanks!

I am getting this now:

```
 CONFIRMED message retries : 3                                            

 Adaptive data  rate (ADR) - Enabled 
[DBG ][LSTK]: Initiating OTAA
[DBG ][LSTK]: Send Join-request..
[DBG ][LMAC]: Next Channel Idx=2, DR=5

 Connection - In Progress ...

[DBG ][LMAC]: Next Channel Idx=2, DR=3
[DBG ][LMAC]: Next Channel Idx=0, DR=2
[DBG ][LMAC]: Next Transmission in 14057 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=2
[DBG ][LMAC]: Next Channel Idx=0, DR=1
[DBG ][LMAC]: Next Transmission in 30208 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=1
[DBG ][LMAC]: Next Channel Idx=0, DR=0
[DBG ][LMAC]: Next Transmission in 74658 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=0
[DBG ][LMAC]: Next Channel Idx=0, DR=5
[DBG ][LMAC]: Next Transmission in 140009 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=5
[DBG ][LMAC]: Next Channel Idx=0, DR=4
[DBG ][LMAC]: Next Channel Idx=0, DR=3
[DBG ][LMAC]: Next Transmission in 4406 ms
[DBG ][LMAC]: Next Channel Idx=1, DR=3
[DBG ][LMAC]: Next Channel Idx=0, DR=2
[DBG ][LMAC]: Next Transmission in 14057 ms
[DBG ][LMAC]: Next Channel Idx=2, DR=2
[DBG ][LMAC]: Next Channel Idx=0, DR=1
[DBG ][LMAC]: Next Transmission in 30208 ms
[DBG ][LMAC]: Next Channel Idx=0, DR=1
[DBG ][LMAC]: Next Channel Idx=0, DR=0
[DBG ][LMAC]: Next Transmission in 74658 ms
[DBG ][LMAC]: Next Channel Idx=0, DR=0

 OTAA Failed - Check Keys ̌

```

It seems I am just out of range of the GW in the area...

Electrobub, 2018-05-29:
I managed to setup a single channel GW for testing. 
And it ended up detecting this mbed node (on TTN and the GW logs but just the data wasn't confirmed properly - I assume that is due to being limited to a single channel and this mbed code not sending / receiving an TX / RX on the same freq)

hasnainvirk, 2018-06-01:
@Electrobub Single channel gateways will always have issues, not only because of duty cycling, but also because you are not standard compliant by any stretch of imagination. Standard says, every base station needs to have 3 channels dedicated for Joining process. A base station must fulfil this criteria.

Electrobub, 2018-06-09:
Yes of course and thanks for the replies! I can safely the original issue was resolved.

closed 2018-06-09 by Electrobub