ARMmbed/mbed-os-example-lorawan #78: Export from Online IDE to make_gcc_arm failed

peksaa01, 2018-06-04:
Export works but compiling failed. I try first compile it and then try to deploy as compile recommended:

peksaa01@E109947 MINGW64 ~/oob2/mbed-os-example-lorawan_gcc_arm_xdot_l151cc/mbed-os-example-lorawan
$ mbed compile -m XDOT_L151CC -t GCC_ARM
[mbed] ERROR: The mbed tools were not found in "C:\Users\peksaa01\git\oob2\mbed-os-example-lorawan_gcc_arm_xdot_l151cc\mbed-os-example-lorawan".
[mbed] ERROR: Run `mbed deploy` to install dependencies and tools.
---

peksaa01@E109947 MINGW64 ~/oob2/mbed-os-example-lorawan_gcc_arm_xdot_l151cc/mbed-os-example-lorawan
$ mbed deploy
[mbed] WARNING: Program "mbed-os-example-lorawan" in "C:\Users\peksaa01\git\oob2\mbed-os-example-lorawan_gcc_arm_xdot_l151cc\mbed-os-example-lorawan" does not use source control management.
[mbed] WARNING: To fix this you should use "mbed new ." in the root of your program.
---
[mbed] ERROR: Library reference "mbed-lora-radio-drv.lib" points to a folder "C:\Users\peksaa01\git\oob2\mbed-os-example-lorawan_gcc_arm_xdot_l151cc\mbed-os-example-lorawan\mbed-lora-radio-drv", which is not a valid repository.
[mbed] ERROR: You can remove the conflicting folder manually and use "mbed deploy" to import the missing libraries
[mbed] ERROR: You can also remove library reference "C:\Users\peksaa01\git\oob2\mbed-os-example-lorawan_gcc_arm_xdot_l151cc\mbed-os-example-lorawan\mbed-lora-radio-drv.lib" and use "mbed sync" again.

janjongboom, 2018-06-04:
After exporting you shouldn't build with Mbed CLI, but with the IDE you exported to (e.g. `make`).

AnttiKauppila, 2018-06-04:
I verified and make will compile. Closing this

closed 2018-06-04 by AnttiKauppila