ARMmbed/mbed-os-example-lorawan #80: LoRa configuration for Network ID name and passphrase

imsherlock, 2018-06-06:
Are there configuration options for setting a Network ID name and passphrase to connect to a Gateway device?

janjongboom, 2018-06-06:
The name/passphrase is a Multi-Tech proprietary thing. I assume there is some conversion from the Network ID and passphrase to session keys. Maybe these are documented by Multi-Tech...

hasnainvirk, 2018-06-11:
@imsherlock You don't need a a netid or passphrase. As Jan mentioned its a proprietary thing.
In case of OTAA, netid comes down as part of Join Accept message.  
In case of ABP, Mbed OS LoRaWAN stack creates a network ID deduced your device address.
The device address  however you need to configure in case of ABP and that's how you do it:
```json
"lora.device-address": "0x12345678"
```

hasnainvirk, 2018-06-11:
@imsherlock If this answers your question, please close the ticket.

hasnainvirk, 2018-06-20:
Closing this for inactivity. Please reopen if something is unclear.

closed 2018-06-20 by hasnainvirk