ARMmbed/mbed-os-example-lorawan #82: How to configure to send unconfirmed messages ?

HarishkumarBringiramakannu, 2018-06-12:
When I send some data from device to server, I get the acknowledgement back. But I don't want to get the acknowledgement. Could you say how to configure this ?

HarishkumarBringiramakannu, 2018-06-12:
Hello Everyone,
I am using B-L072Z-LRWAN1 device with online mbed compiler. But for some of my programs,I get error as ( Error: Target "DISCO_L072CZ_LRWAN1" is not recognized ). Do anyone knows why ? 
Currently I am working with DS18B20 1-wire temperature sensor. If anyone has the code, you can suggest me, so I can try.

hasnainvirk, 2018-06-13:
@HarishkumarBringiramakannu I can import the example to online compiler and build it successfully. Here is what I did:

1) https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-lorawan/ went to this link.
2) Click 'import to online compiler'
3) Right click on the example app and click 'update all'
4) Click at the board button in the upper right corner and select DISCO_L072CZ_LRAWAN1 as the platform. If you don't see that in your list, click 'Add board'. Search for DISCO_L072CZ_LRAWAN1 on the page and click 'Add to online compiler. '
5) Click 'compile'.

Can you please verify that you can build following the above mentioned steps ?

HarishkumarBringiramakannu, 2018-06-13:
@hasnainvirk  Thank you so much for your reply ! I am using this program from yesterday. Yeah it's running good. But for some of my other program the error shows like what I mentioned.
For Example : I am using DS18B20 1-wire temperature sensor, but when I run a code it shows as ( Error: Target "DISCO_L072CZ_LRWAN1" is not recognized ). If I update my mbed library , it shows some other different errors. So do you have idea how to clear this ?

hasnainvirk, 2018-06-13:
@HarishkumarBringiramakannu Honestly speaking, I have never seen such an error message before. If you are following the instructions properly, you should not see this sort of message. Okay, let's try to get to the bottom of this.  Tell me exactly what steps are you taking to build the example application ?

HarishkumarBringiramakannu, 2018-06-13:
@hasnainvirk I am using the program in this link https://os.mbed.com/users/feabhas/code/DS18B20/ . 
 I give as import to compiler. My board is already selected. So if I give save&compile, I get the error what I mentioned.
But before few days , some told me my mbed online compiler would have been crashed, it might also be reason. But even now some of my programs are running. So I don't know why.
You clear with my explaination ? Please say me if you don't understand.

hasnainvirk, 2018-06-13:
@HarishkumarBringiramakannu In my humble opinion this is the wrong place to open this issue. The example you are trying to build is for Mbed OS 2 and for some reason yes, there seems to be some issue with this platform for Mbed OS 2 builds. I have escalated that. But I still recommend you to open a general issue on Mbed OS for this. You have opened issue for the lorawan example which doesn't have a connection to the issue you are facing. If you can't open the ticket on Mbed OS, please let me know and I will open it on your behalf.

hasnainvirk, 2018-06-13:
@HarishkumarBringiramakannu If you are time bound, and you wish to test your sensor with the lorawan example, try copying only DS18B20 related code to the lorawan example and instead of dummy sensor, use  your original sensor. With Mbed OS 5 , I can confirm that DISCO_L072CZ_LRWAN1 is working. 
We have started investigation for Mbed OS 2 and we will have a solution integrated into next patch release of Mbed OS 2. Thank you for pointing this out to us.

jmarcelino, 2018-06-13:
@HarishkumarBringiramakannu to compile for the board follow this guide, it's the best way:

https://nicbkw.com/stmicro-nucleo-disco-l072cz-lrwan1-quickstart-mbed-os/

For DS18B20 I used a slightly modified version of this

https://os.mbed.com/users/tejaskale/code/DS1820/

But I agree this is the wrong place to ask these questions, feel free to connect on MBED forum: https://os.mbed.com/users/jmarcelino/

HarishkumarBringiramakannu, 2018-06-13:
Hi @hasnainvirk Thanks for your patience. I will try the following what you said and I will get back to you if I have any doubts.

janjongboom, 2018-06-18:
@HarishkumarBringiramakannu This board is not compatible with Mbed OS 2, so to use an Mbed 2 example (like the DS1820 example you found) you'll need to switch to Mbed OS 5. This is very easy and 95% of the time the example will work out of the box. See [migrating in the Mbed OS 5 docs](https://os.mbed.com/docs/v5.8/tutorials/migrating.html).

HarishkumarBringiramakannu, 2018-06-18:
Hi @janjongboom  I am using mbed online compile. So it always has the updated version. Is there any other way to run my program. My problem is, whatever I do, I get error in the header files. So could you help me through this ?

janjongboom, 2018-06-19:
@HarishkumarBringiramakannu I see that that page is indeed lacking instructions on how to do it in the Online Compiler (created bug [here](https://github.com/ARMmbed/mbed-os-5-docs/issues/597)). In the Online Compiler you do:

1. Right click on `mbed`, select *Delete...*.
1. Right click on your project, select *Import Library > From URL*.
1. Enter `https://github.com/armmbed/mbed-os`.

And done, now the application is updated to latest version of Mbed OS. Hope this helps.

HarishkumarBringiramakannu, 2018-06-19:
Hi @janjongboom ,
Thankyou for your link. It was very helpful. But now I am struck with this error "Error: Library name 'platform' is not unique (defined in '/tmp/chroots/ch-2619ff6b-a6f9-4624-81ff-1ac9c3c14ea0/extras/mbed/TARGET_DISCO_L072CZ_LRWAN1/mbed_lib.json' and '/tmp/chroots/ch-2619ff6b-a6f9-4624-81ff-1ac9c3c14ea0/src/mbed-os/platform/mbed_lib.json')"

when I was searching in internet, many got this error but they don't know the solution, do you know how to clear this ? could you help me out through this ?

hasnainvirk, 2018-06-27:
@HarishkumarBringiramakannu  Can you please open an issue on Mbed OS in general and close it from here as it is not related to this project ?

closed 2018-06-27 by HarishkumarBringiramakannu