ARMmbed/mbed-os-example-lorawan #93: Hard fault on CN470

janjongboom, 2018-08-14:
Sending anything over the CN470 channel plan yields a hard fault. Confirmed by Ronny on the K64F.

```

.[2K.[90m[DBG ][LSTK]: Initializing MAC layer.[0m

 Mbed LoRaWANStack initialized 

 CONFIRMED message retries : 3 

 Adaptive data  rate (ADR) - Enabled 

.[2K.[90m[DBG ][LSTK]: Initiating ABP.[0m

.[2K.[90m[DBG ][LSTK]: Frame Counters. UpCnt=0, DownCnt=0.[0m

.[2K.[90m[DBG ][LSTK]: ABP connection OK..[0m

 Connection - In Progress ...

 Connection - Successful 

 Dummy Sensor Value = 2.1 

.[2K.[39m[INFO][LMAC]: RTS = 25 bytes, PEND = 0, Port: 15.[0m

.[2K.[90m[DBG ][LMAC]: Frame prepared to send at port 15.[0m

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000000
R1   : 33323130
R2   : 008F218D
R3   : F82C4992
R4   : 0000005F
R5   : 00000000
R6   : 00000001
R7   : 00000008
R8   : 00000060
R9   : 00000000
R10  : 0000011D
R11  : 20001148
R12  : 2000399C
SP   : 20003200
LR   : 0000EB5F
PC   : 0000EB74
xPSR : 41000000
PSP  : 20003198
MSP  : 2002FFD8
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000004
UFSR : 00000000
DFSR : 00000008
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x11871
Error Value: 0xEB74
Current Thread: Id: 0x200033B8 Entry: 0x14D39 StackSize: 0x1000 StackMem: 0x200023B8 SP: 0x2002FFA0 
-- MbedOS Error Info --
```

imi415, 2018-08-15:
Can also confirm this issue on NUCLEO_L476RG with OTAA activation.

cmonr, 2018-08-15:
@janjongboom Compiler version?

@imi415 GCC 8 is not supported at this time.
https://os.mbed.com/docs/latest/tools/configuring-mbed-cli.html

imi415, 2018-08-16:
@cmonr Build with GCC 6 2017-q2 released by ARM produced the same result.

imi415, 2018-08-16:
This issue occurs when LoraMac trying to send a packet, as OTAA needs to send join request at initial connecting step.
```
[DBG ][LSTK]: Initializing MAC layer

 Mbed LoRaWANStack initialized

 CONFIRMED message retries : 3

 Adaptive data  rate (ADR) - Enabled 
[DBG ][LSTK]: Initiating OTAA
[DBG ][LSTK]: Sending Join Request ...
[DBG ][LMAC]: Frame prepared to send at port 0

++ MbedOS Fault Handler ++
...
```

imi415, 2018-08-16:
Solved: Tracing back call stack found that LoRaPHY.cpp predefined channel enabled array with length 16, however CN470 has 96. Will make a pr later by override that method.

hasnainvirk, 2018-08-16:
The issue is a missing override for set_next_channel() function. The system is picking up a default implementation of set_next_channel() from LoRaPHY. Will send a patch today.

AnttiKauppila, 2018-09-13:
@janjongboom Seems this is fixed, so closing. Reopen if the problem still exists

closed 2018-09-13 by AnttiKauppila

nimsnaik, 2019-05-29:
I am getting the same hard fault issue when using sx1276 shield and stm32 L152RE board.

I am using the online mbed tools to compile, how can i fix this?