ARMmbed/mbed-os-example-lorawan #103: Not receiving downlinks when using low power timers (STM32F070 / SX1276)

PietKnaak, 2018-10-10:
I'm using a STM32F070 board with a SX1276 radio. When using the normal ticker all (joins, uplinks and downlinks) seems to work fine. When we switch to low power tickers (and enable the use-lowpower-timer-ticker in the events library config) the downlinks stop to work.

Has the stack been tested with low power tickers this or other processors? 

For debugging, can you give us some pointers where to start digging?

janjongboom, 2018-10-10:
@hasnainvirk

hasnainvirk, 2018-10-10:
@PietKnaak Are you building with RTOS or non-RTOS ? Have you twiddled radio driver timers also to low power timeout ?

hasnainvirk, 2018-10-10:
@PietKnaak I have tested Deep Sleep with K64F + SX1276 and it worked as it should.
Only issue I found was a buggy trace as the K64F have a buggy Deep Sleep implementation. 
Things you need (if using RTOS): 
i)  MBED_TICKLESS must be defined in either targets.json or mbed_app.json (make sure that your platform supports it)
ii) Make sure that LPTICKER is defined for your platform.
iii) Following code snippet for radio driver:
```C
#if DEVICE_LPTICKER
#include "LowPowerTimeout.h"
#define ALISAS_LORAWAN_TIMER    mbed::LowPowerTimeout
#else
#include "Timeout.h"
#define ALISAS_LORAWAN_TIMER    mbed::Timeout
#endif

// find it in private variablles
ALISAS_LORAWAN_TIMER tx_timeout_timer;
```

That is it.  'use-lowpower-timer-ticker' doesn't have any effect for RTOS.

For tracing that how much time you are spending in Deep Sleep:
```C
#include "mbed_stats.h"
static mbed_stats_cpu_t stats;

void my_func()
{
    mbed_stats_cpu_get(&stats);

    printf("Uptime %" PRId64 "\n", stats.uptime);
    printf("idle time %" PRId64 "\n", stats.idle_time);
    printf("sleep time %" PRId64 "\n", stats.sleep_time);
    printf("depp sleep time %" PRId64 "\n", stats.deep_sleep_time);
}
```

Here is the example output:

```
Uptime 595471710
idle time 587811371
sleep time 203494
depp sleep time 5876078770
```

I don't have the platform you specified at my disposal, so you should make sure that it is capable of going into Deep Sleep.

PietKnaak, 2018-10-10:
@hasnainvirk thank you for replying. We use mbed5.8 without RTOS. My mbed ignore looks like:

`mbed-os/rtos/*
mbed-os/features/filesystem/*
mbed-os/features/cellular/*
mbed-os/features/frameworks/*
mbed-os/features/net/*
mbed-os/features/netsocket/*
mbed-os/features/storage/*
mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/*
mbed-os/features/nanostack/*`

I'm using these radio drivers: https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers#c4dd25eec9ae30f9532e498ea7e2cacd758c733e

> Have you twiddled radio driver timers also to low power timeout ?

Yes, but only with the LSI which results in downlinks on a high Data Rate, we are now modifying the board to use the LSE to see if that works more reliable.

hasnainvirk, 2018-10-10:
@PietKnaak I have just tried with latest Mbed OS without RTOS.

.mbedignore will look like this:

```
mbed-os/rtos/* 
mbed-os/components/*
mbed-os/features/filesystem/* 
mbed-os/features/cellular/* 
mbed-os/features/net/* 
mbed-os/features/netsocket/* 
mbed-os/features/storage/* 
mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/* 
mbed-os/features/nanostack/*
mbed-os/features/device_key/*
mbed-os/features/TEST_APPS/*
mbed-os/features/TESTS/*
mbed-os/features/UNITTESTS/*
mbed-os/features/frameworks/greentea-client/*
mbed-os/features/frameworks/mbed-client-cli/*
mbed-os/features/frameworks/mbed-coap/*
mbed-os/features/lwipstack/*
mbed-os/features/frameworks/unity/*
mbed-os/features/frameworks/utest/*
```
Take this PR from radio driver https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers/pull/31

mbed_app.json settings:

```json
 "platform.cpu-stats-enabled": true,
  "events.use-lowpower-timer-ticker": true,
```
Everything works nicely, except the trace being bonkers, e.g., 
```
Uptime 265087921
idle time 263695169
sleep time 0
0epp sleep time 263695169
```
I would recommend that you should use latest Mbed OS (patch release coming today).
There are significant improvements in the Deep Sleep performance.

Size of the build:

10K RAM, 87K Flash. You can save at least 4-5K of RAM by using newlib-nano which would then result in RAM footprint of about ~5K.

PietKnaak, 2018-10-10:
@hasnainvirk using an external LSE (low frequency xtal) and patching the timers as suggested in your PR solved the issue.

I will update mbed-os tomorrow.

Thanks a lot for your quick support!

janjongboom, 2018-10-10:
@hasnainvirk @PietKnaak Great!

hasnainvirk, 2018-10-11:
@PietKnaak Pleasure is all mine. Can you please go ahead and close the issue if you are satisfied ?

closed 2018-10-12 by PietKnaak