ARMmbed/mbed-os-example-lorawan #105: ACK_TIMEOUT ELAPSE and ERROR CODE -1010

jefdoggy, 2018-10-12:
I am using X-NUCLEO-53L0A1 and NUCLEOF401RE with the SX1272 shield.
I try to send the proximity sensor data to the gateway but it keeps showing ack_timeout and finally a send error code -1010.
[image]
[image]
[image]
[image]

I only added the sensor data into the send message function
[image]

janjongboom, 2018-10-12:
Does it work if you remove all sensor code and *physically* disconnect the sensor board? It could be that your sensors are wired to pins that are also used by the LoRa shield - although doesn't look like it from first glance.

Note that you're initializing objects on every call to `send_message` now so you're leaking memory and probably introducing bugs in the sensor code.

jefdoggy, 2018-10-12:
@janjongboom thanks for your reply.

I've changed the pins for the sensor so there will be no collision. 

Actually, even if I remove the sensor code and the sensor, and simply run the example which with the dummy sensor code, the trace log shows similar.

I can join the network successfully, and the sensor data can be received from the LORA gateway, but only can receive no more than twice. After that, the board keeps sending empty message. It seems it will not entering the send_message loop after two times.

janjongboom, 2018-10-12:
@hasnainvirk Could this be an uncalibrated crystal?

hasnainvirk, 2018-10-12:
@jefdoggy It's hard to pin down what is going on at your end. It could be uncaliberated crystals on board. However, I can notice something peculiar from your trace. You are probably located in China or US and your PHY layer is picking up channels from full range. Are you sure that the base station in your vicinity is capable of handling these many channels ? Most base stations in production can support only 8 channels, and if you haven't configured FSB mask for your PHY, you may be swinging into channels where nobody is listening to you.

janjongboom, 2018-10-24:
@jefdoggy Those are warnings, not errors.

hasnainvirk, 2018-10-25:
@jefdoggy How many channels does your base station supports ?

jefdoggy, 2018-10-25:
@hasnainvirk it supports 8 channels. I updated the code to the latest one and it works fine now though still not find where the problem is. Thanks for your help anyway.