ARMmbed/mbed-os-example-lorawan #111: Issue: error on downlink

adustm, 2018-11-27:
Hello lorawan experts,
Could someone have a look at the following issue, please ?
https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-lorawan/issues/6

cc @cdupaty

Thanks in advance

janjongboom, 2018-11-27:
@hasnainvirk @AnttiKauppila

hasnainvirk, 2018-11-27:
@adustm The issue is that they are sending on a Port which is not being listened at. 
They are probably using an incarnation of receive() API which listens to a specific Port with specific flags.
If they wish to receive on any port and any flags, use this api:

```
int16_t LoRaWANInterface::receive(uint8_t *data, uint16_t length, uint8_t &port, int &flags)
```

hasnainvirk, 2019-01-02:
@adustm Please close the ticket if you believe the issue is resolved, otherwise let us know how we can assist you ?

hasnainvirk, 2019-02-13:
Closing issue because of inactivity, answer was provided to the reporter.

closed 2019-02-13 by hasnainvirk