ARMmbed/mbed-os-example-lorawan #112: Disabling ADR

DavidNickalls, 2018-11-30:
I'm trying to disable the adaptive data rate to asses the performance of the network with a fixed spreading factor. 
To do this I added a line in the json file: '"lora.adr-on": false,' in  "target_overrides".
In the main i added 'lorawan.disable_adaptive_datarate()' 
Then i set the new data rate using 'lorawan.set_datarate(DR_#)'
each return 'LORAWAN_STATUS_OK'
I have experimented setting different DRs.
On The Things Network I get 5 activations, each of these has a different data_rate, starting with a data_rate of SF7BW125, with the final having a data_rate = SF11BW125.
After activation all packets are sent with the final data_rate.
From this I assume the ADR has not successfully been disabled, any advice would be much appreciated!

janjongboom, 2018-12-06:
ADR is a network feature, so this has nothing to do with joining the network. In OTAA mode the spreading factor is set by the network stack after a successful join. If you want to override it you can set the spreading factor manually after the `CONNECTED` event happens (just tested this on Mbed OS 5.10.2).

As far as I know it's not possible to set the join data rate, and it will always start at SF7 and go up (in Europe). But maybe @hasnainvirk knows a way.

Typically for range tests I use ABP, and just fix the data rate to SF12. Saves time re-joining the network if you ever accidentally reset the device.

hasnainvirk, 2018-12-07:
@DavidNickalls It is not possible to set data rate for Join Messages because the specification binds us to follow a specific progression. Join Requests always go our starting with lowest SF, i.e., SF7 and increase if the Join Accept message is not received (i.e., increase range). The algorithm we follow is given in chapter 7 of the specification. 
After joining, you can set the data rate for the device if the ADR is disabled. I am trying to understand what you mean by 5 activations. Can you please elaborate ? The stack can have only one session (only one activation).

ivoknutsel, 2019-01-03:
> Typically for range tests I use ABP, and just fix the data rate to SF12. Saves time re-joining the network if you ever accidentally reset the device.

How do I pin the SF (and BW) ? I was hoping to find a parameter in the .json config file.

hasnainvirk, 2019-01-03:
@ivoknutsel By default the stack uses ADR. So if you wish to pin down SF:
i) Disable ADR https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/LoRaWANInterface.h#L62
ii) Set your desired DR/SF https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/LoRaWANInterface.h#L60

Bandwidth is automatically selected based on your DR according to the spec.

janjongboom, 2019-01-04:
@ivoknutsel For ABP just set the data rate and disable ADR, and that's it.

For OTAA set the data rate in the `CONNECTED` event handler.

hasnainvirk, 2019-02-13:
Closing because of inactivity. Answer was provided to the reporter. Please reopen if there is still something unanswered.

closed 2019-02-13 by hasnainvirk