ARMmbed/mbed-os-example-lorawan #122: Program size

x893, 2019-02-06:
Looks good but how to reduce program size ?

| Module                     |       .text |    .data |      .bss |
|----------------------------|-------------|----------|-----------|
| [fill]                     |     204(-4) |   12(+0) |    42(+0) |
| [lib]\c.a                  |   25193(+0) | 2472(+0) |    89(+0) |
| [lib]\gcc.a                |    4820(+0) |    0(+0) |     0(+0) |
| [lib]\m.a                  |     576(+0) |    0(+0) |     0(+0) |
| [lib]\misc                 |     252(+0) |   16(+0) |    28(+0) |
| main.o                     |    1792(-4) |    4(+0) |  4084(+0) |
| mbed-lora-radio-drv\SX1272 |     495(+0) |    0(+0) |     0(+0) |
| mbed-lora-radio-drv\SX1276 |    8367(+0) |    0(+0) |     0(+0) |
| mbed-os\cmsis              |    1033(+0) |    0(+0) |    84(+0) |
| mbed-os\drivers            |    2011(+0) |    4(+0) |   152(+0) |
| mbed-os\events             |    1496(+6) |    0(+0) |     0(+0) |
| mbed-os\features           |  30706(-22) |    4(+0) |     0(+0) |
| mbed-os\hal                |    3388(+0) |    8(+0) |   247(+0) |
| mbed-os\platform           |    3905(+0) |  260(+0) |   268(+0) |
| mbed-os\rtos               |   11118(+0) |  168(+0) |  5973(+0) |
| mbed-os\targets            |   13735(+0) |    4(+0) |  1017(+0) |
| trace_helper.o             |       2(+0) |    0(+0) |     0(+0) |
| Subtotals                  | 109093(-24) | 2952(+0) | 11984(+0) |
Total Static RAM memory (data + bss): 14936(+0) bytes
Total Flash memory (text + data): 112045(-24) bytes`

janjongboom, 2019-02-07:
@x893 If you have size issues you can use two (three) major tricks: 1) remove RTOS, 2) build with newlib-nano, 3) use minimal-printf. See https://github.com/janjongboom/mbed-os-example-blinky-no-rtos

x893, 2019-02-07:
I try 2) and 3) and reduce to 104 KB.  I think you know more about internals for mbed-os/features and other mbes-os parts.

Thanks

janjongboom, 2019-02-07:
Well, try 1. It will get rid of an additional 40K flash.

evandavey, 2019-03-29:
Hi,
I am using the exact mbed ignore in https://github.com/janjongboom/mbed-os-example-blinky-no-rtos and the same tiny.json build profile.  I can see a join accept in the TTN console but the device never receives a CONNECT event (or any other event) and gets stuck at "Connection - In Progress ...".

janjongboom, 2019-03-29:
@evandavey Can you enable traces and post a new issue with the trace log? See https://github.com/ARMmbed/mbed-os-example-lorawan#optional-adding-trace-library.

evandavey, 2019-03-29:
OK will do tomorrow my time (currently evening).

janjongboom, 2019-03-29:
@evandavey Just FYI, I've tested the example on DISCO-L475VG-IOT01A1 with SX1276 radio with .mbedignore and tiny profile and it connects fine for me; interested to see the traces.

evandavey, 2019-03-29:
@janjongboom good to know, thanks.  This is a custom PCB using a Murata Type ABZ module - I had it connecting using the stock example but now that seems to be throwing a 
HAL_RCC_OscConfig ERROR so not sure what's up with that (using the cloud compiler and target DISCO-L072CZ-LRWAN1 as MTB_MURATA_ABZ doesn't work).  The example above compiled offline with GCC_ARM and MTB_MURATA_ABZ as the target.

evandavey, 2019-04-10:
Looks like there is now a 'bare-metal' mode for mbed-os 5.12.0 https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-5.12.0.  How would I use that for this project?

kluszkiewicz, 2019-05-27:
@janjongboom 
> @x893 If you have size issues you can use two (three) major tricks: 1) remove RTOS, 2) build with newlib-nano, **3) use minimal-printf. See https://github.com/janjongboom/mbed-os-example-blinky-no-rtos**

Solution 3. (disabling mbed features in .mbedignore) works fine for mbed-os 5.11.*, but does not work with ver. 5.12.* - plenty of dependencies to header files from disabled modules (nanostack, cellular,...) like e.g.:

'
Compile [  1.6%]: rf_configuration.c
[Fatal Error] rf_configuration.c@16,44: nanostack/platform/arm_hal_phy.h: No such file or directory
[ERROR] .\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver\source\rf_configuration.c:16:44: fatal error: nanostack/platform/arm_hal_phy.h: No such file or directory
 #include "nanostack/platform/arm_hal_phy.h"
'
Is it possible still to disable mbed features in mbed-os version >=5.12?

0xc0170, 2019-05-27:
@kluszkiewicz if you use baremetal profile, it should do the trick. If you still have an issue, file a new issue on Mbed OS repository with details

kluszkiewicz, 2019-05-27:
> 
> 
> @kluszkiewicz if you use baremetal profile, it should do the trick. If you still have an issue, file a new issue on Mbed OS repository with details

But what if I want to disable some features like features/cellular, lwip, ...? 
Can i control mentioned features using baremetal option?

kluszkiewicz, 2019-05-27:
> 
> 
> @kluszkiewicz if you use baremetal profile, it should do the trick. If you still have an issue, file a new issue on Mbed OS repository with details

I've reported issue on Mbed OS repo:
https://github.com/ARMmbed/mbed-os/issues/10678

AnttiKauppila, 2019-05-27:
The bug is valid and is going to be fixed. I have commented to issue how you can proceed with this.

closed 2019-09-13 by AnttiKauppila