ARMmbed/mbed-os-example-lorawan #124: What to use for "lora.application-key"?

ssctx66, 2019-02-23:
Using a Multitech mDot and Conduit, I get the following output from this example:

  ```
  Mbed LoRaWANStack initialized
  CONFIRMED message retries : 3
  Adaptive data  rate (ADR) - Enabled
  Connection - In Progress ...
  OTAA Failed - Check Keys

```
Per https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-lorawan/file/dfce67eca082/mbed_app.json/, I need to add `Device EUI`, `Application EUI` and `Application Key' to mbed_app.json  I am using the node ID (as shipped with mDot) for "Device EUI" and the "hexified" version of our LoRaWAN network name (hex equivalent obtained with AT commands) for "Application EUI."  I have no idea what to use for "Application Key," though.

How do I figure out an acceptable 16-byte application key?

AnttiKauppila, 2019-02-25:
@ssctx66 You need to have a matching EUIs and App key with Conduit AEP:
http://www.multitech.net/developer/software/aep/lorawan-aep/key-management/

ssctx66, 2019-02-25:
Thanks, @AnttiKauppila.  Still a bit confused, though.

Do you mean the AppEUI and AppKey on the mDot has to match those on the Conduit?  If so, I know that but still do not know how to derive the AppKey.

If you mean instead that the AppEUI and AppKey have to match one another (ie, be the same), lines 37-38 in the mbed_app.json example (in "File contents..." showing actual example code at the bottom of the link I provided upthread), seem to indicate otherwise:
```
"lora.application-eui": "{ 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }",
"lora.application-key": "{ 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }"
```
This implies to me AppEUI is 8 bytes whereas AppKey is 16 bytes.

janjongboom, 2019-03-01:
I don't have an AEP Conduit, but in https://assets.lairdtech.com/home/brandworld/files/LoRaWAN%20Keys%20and%20IDs%20Overview.pdf it seems that you can set the Network Key to use a proper key.

maiorfi, 2019-03-16:
Hi.

I'm having same issue with Multitech Conduit (AEP v1.7.2).

Using an MDOT device and leaving default "all-zeroes" configuration for lora.application-eui and lora.application-key, I get following output (with mbed_trace enabled):

```
Mbed LoRaWANStack initialized

 CONFIRMED message retries : 3

 Adaptive data  rate (ADR) - Enabled
[DBG ][LSTK]: Initiating OTAA
[DBG ][LSTK]: Sending Join Request ...
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: TX: Channel=1, TX DR=5, RX1 DR=5

 Connection - In Progress ...
[DBG ][LSTK]: Transmission completed
[DBG ][LMAC]: RX1 slot open, Freq = 868100000
[DBG ][LMAC]: RX2 slot open, Freq = 869525000
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: TX: Channel=0, TX DR=4, RX1 DR=4
[DBG ][LSTK]: Transmission completed
[DBG ][LMAC]: RX1 slot open, Freq = 868300000
[DBG ][LMAC]: RX2 slot open, Freq = 869525000
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: DC enforced: Transmitting in 5264 ms
[DBG ][LMAC]: TX: Channel=1, TX DR=3, RX1 DR=3
[DBG ][LSTK]: Transmission completed
[DBG ][LMAC]: RX1 slot open, Freq = 868100000
[DBG ][LMAC]: RX2 slot open, Freq = 869525000
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: DC enforced: Transmitting in 14494 ms
[DBG ][LMAC]: TX: Channel=1, TX DR=2, RX1 DR=2
[DBG ][LSTK]: Transmission completed
[DBG ][LMAC]: RX1 slot open, Freq = 868300000
[DBG ][LMAC]: RX2 slot open, Freq = 869525000
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: DC enforced: Transmitting in 30740 ms
[DBG ][LMAC]: TX: Channel=1, TX DR=1, RX1 DR=1
[DBG ][LSTK]: Transmission completed
[DBG ][LMAC]: RX1 slot open, Freq = 868300000
[DBG ][LMAC]: RX2 slot open, Freq = 869525000
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: DC enforced: Transmitting in 76048 ms
...
 OTAA Failed - Check Keys
```

On Gateway side, I have configured Conduit Key Management section as shown here:
https://www.dropbox.com/s/271cfsv41zaz8lv/issue_lorawan_mbedos_1.png?dl=0

Also, looking at LoRaWAN diagnostics on Conduit side, I can see joining has been done successfully.

Any suggestion?

mjohngreene, 2019-03-16:
Yes, @janjongboom, we can set the Network Key/AppKey (see example of what the AEP Conduit's screen looks like in @maiorfi's linked screenshot above).  Whereas the AEP Conduit allows users to enter actual text for both Network ID/AppEUI and Network Key/AppKey, though, the mbed-os-example-lorawan appears to only accept hex.  We can figure out what the hex equivalent is for our Network ID/AppEUI with AT commands on an mdot.  However, we are still not sure what to use for the Network Key/AppKey.  And making one up does not seem to work.

mjohngreene, 2019-03-31:
Finally got it working with some help from the following: 

http://www.multitech.net/developer/forums/topic/network-id-appeui-transformation/ 

Also, needed to add a "target override" to mbed_app.json for `"lora.public-network": false,` was necessary as we run private LoRaWANSs.

closed 2019-03-31 by mjohngreene