ARMmbed/mbed-os-example-lorawan #126: Error code -1014

boonsanti, 2019-03-01:
Got send() - Error code -1014 when I enable AnalogIn like below code.If I take it out then compile and download to board. Eveything run ok.

AnalogIn ai0(A0);

Library added:
CayenneLPP

Toolchain:
gcc-arm-none-eabi-6-2017-q2-update

OS:
Ubuntu 18.04

janjongboom, 2019-03-01:
On the SX1272 and SX1276 shields pin `A0` is mapped to `NRESET` on the radio. So if you're declaring a sensor there there will be a conflict, and thus the radio cannot be driven.

boonsanti, 2019-03-01:
> On the SX1272 and SX1276 shields pin `A0` is mapped to `NRESET` on the radio. So if you're declaring a sensor there there will be a conflict, and thus the radio cannot be driven.

Sorry @janjongboom I forgot to tell you the platform I run on is ST DISCO-L072CZ-LRWAN1.

hasnainvirk, 2019-03-01:
@boonsanti You can't use A0 or PA0 on DISCO-L072CZ-LRWAN1 either. This pin is mapped to SYS wakeup 1 for debugging.

boonsanti, 2019-03-01:
> @boonsanti You can't use A0 or PA0 on DISCO-L072CZ-LRWAN1 either. This pin is mapped to SYS wakeup 1 for debugging.

@hasnainvirk I change to A2 but got same error.So cannot use analog pin during lora chip active right?

janjongboom, 2019-03-04:
@boonsanti Yes, you can, but you should find one that is not used by the radio. Looking at the pinout for the DISCO-L072CZ-LRWAN1 board should give you some info.

boonsanti, 2019-03-04:
Finally,I decide to use ADC I2C module with DISCO-L072CZ-LRWAN1 for my project.Thanks all! :)

closed 2019-03-04 by boonsanti