ARMmbed/mbed-os-example-lorawan #139: Change the spreading factor

fortudiego, 2019-05-15:
Hi I'm using your code and everything works correctly, but I would like to increase the spreading factor (SF) that I saw to be default to 7. How can I do ??

Thanks for the attention

fortudiego, 2019-05-17:
I will be very grateful if you could send me the right link. I'm not able to find it
Thanking in advance.

AnttiKauppila, 2019-05-17:
See https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/LoRaWANInterface.h#L213
And https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/lorawan_types.h#L354 onwards

AnttiKauppila, 2019-05-17:
Also ADR must be switched off

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila