ARMmbed/mbed-os-example-lorawan #142: Connection error, code = -1014 for US915

rajah-ketos, 2019-05-29:
Hi @AnttiKauppila, 

I am currently working on a LoRa node hardware design. I am using the DISCO_L072CZ_LRWAN1 as a reference. I am looking at using the Mbed version of the stack and potentially even use the firmware update features in the future. I have a private LoRa gateway and I have a few nodes that can connect to it. When I try to connect using the DISCO_L072CZ_LRWAN1 with the Mbed code, I am getting the following error on serial port. 

```
 Mbed LoRaWANStack initialized

 CONFIRMED message retries : 3

 Adaptive data rate (ADR) - Enabled

 Connection error, code = -1014
```
I am able to connect to the same gateway using the semtech stack (https://github.com/Lora-net/LoRaMac-node), so I know its not a hardware or gateway issue.

Digging deeper reveals that the error code maps to LORAWAN_STATUS_CRYPTO_FAIL. I am using US915 region and have set the following in mbed_app.json. What could I be missing? I am on MacOS and using mbedCLI to compile.

"target_overrides": {
"*": {
"platform.stdio-convert-newlines": true,
"platform.stdio-baud-rate": 115200,
"platform.default-serial-baud-rate": 115200,
"lora.over-the-air-activation": true,
"lora.duty-cycle-on": true,
"lora.phy": "US915",
"lora.device-eui": "{ 0x31, 0x32, 0x37, 0x37, 0x73, 0x37, 0x90, 0x05 }",
"lora.application-eui": "{ 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01 }",
"lora.application-key": "{ not typing here }"

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila

rajah-ketos, 2019-09-13:
Turned out to be a compiler issue, I was using GCC to compile. However, when I tried the MBED online compiler, I was able to connect without issues. Since I am on Mac, there is no offline version of the compiler for Mac, so I am stuck with the online compiler for now.

0xc0170, 2019-09-16:
This is strange. I believe both use develop profile so should behave the same.  Anyway, thanks for the latest update 👍

rajah-ketos, 2019-09-16:
The code size (RAM usage) of GCC compile is larger than online online compiler (I believe that it uses the Keil compiler from ARM). I assume that the error is a side effect of some stack overflow (?). In any case, for the same codebase, I can compile using online compiler and GCC and only only the binary from online compiler works.