ARMmbed/mbed-os-example-lorawan #144: Sx1272 Radio Fails to Transmit:

farrenv, 2019-06-05:
I am using the mbed-os-example-lorawan example code with the [NaMote-72 platform](https://os.mbed.com/platforms/NAMote-72/) 

I get the following error:
```
[DBG ][LSTK]: Initializing MAC layer

 Mbed LoRaWANStack initialized

 CONFIRMED message retries : 3

 Adaptive data  rate (ADR) - Enabled
[DBG ][LSTK]: Initiating OTAA
[DBG ][LSTK]: Sending Join Request ...
[DBG ][LMAC]: Frame prepared to send at port 0
[DBG ][LMAC]: TX: Channel=71, TX DR=4, RX1 DR=13

 Connection - In Progress ...
[DBG ][LSTK]: TX Timeout
[ERR ][LSTK]: Joining abandoned: Radio failed to transmit

 Transmission Error - EventCode = 3

 Dummy Sensor Value = 2.1

 send() - Error code -1017
```
My mbed_app.json configuration:
(target is MOTE_L152RC)

```
{
    "config": {
        "lora-radio": {
            "help": "Which radio to use (options: SX1272,SX1276)",
            "value": "SX1272"
        },
        "main_stack_size":     { "value": 4096 },

        "lora-spi-mosi":       { "value": "NC" },
        "lora-spi-miso":       { "value": "NC" },
        "lora-spi-sclk":       { "value": "NC" },
        "lora-cs":             { "value": "NC" },
        "lora-reset":          { "value": "NC" },
        "lora-dio0":           { "value": "NC" },
        "lora-dio1":           { "value": "NC" },
        "lora-dio2":           { "value": "NC" },
        "lora-dio3":           { "value": "NC" },
        "lora-dio4":           { "value": "NC" },
        "lora-dio5":           { "value": "NC" },
        "lora-rf-switch-ctl1": { "value": "NC" },
        "lora-rf-switch-ctl2": { "value": "NC" },
        "lora-txctl":          { "value": "NC" },
        "lora-rxctl":          { "value": "NC" },
        "lora-ant-switch":     { "value": "NC" },
        "lora-pwr-amp-ctl":    { "value": "NC" },
        "lora-tcxo":           { "value": "NC" }
    },
    "target_overrides": {
        "*": {
            "mbed-trace.enable":true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "lora.over-the-air-activation": true,
            "lora.duty-cycle-on": true,
            "lora.phy": "US915",
            "lora.device-eui": "{  }",
            "lora.application-eui": "{ }",
            "lora.application-key": "{  }"
        },
[... 202 lines cut ...]
            "lora-spi-miso":       "SPI_RF_MISO",
            "lora-spi-sclk":       "SPI_RF_SCK",
            "lora-cs":             "SPI_RF_CS",
            "lora-reset":          "SPI_RF_RESET",
            "lora-dio0":           "DIO0",
            "lora-dio1":           "DIO1",
            "lora-dio2":           "DIO2",
            "lora-dio3":           "DIO3",
            "lora-dio4":           "DIO4",
            "lora-dio5":           "NC",
            "lora-rf-switch-ctl1": "NC",
            "lora-rf-switch-ctl2": "NC",
            "lora-txctl":          "ANT_CTX_PA",
            "lora-rxctl":          "ANT_CRX_RX",
            "lora-ant-switch":     "NC",
            "lora-pwr-amp-ctl":    "NC",
            "lora-tcxo":           "RF_TCXO_EN"
        }
    },
    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_lora_config.h\""]
}

```

Any suggesstions what the problem could be?  Thanks

loverdeg-ep, 2019-06-05:
@snelson-senet, @hasnainvirk

farrenv, 2019-06-05:
I was using the incorrect of two SPI ports

farrenv, 2019-06-05:
The example still does not work out-of-the-box.

farrenv, 2019-06-05:
The device connected after a power cycle.

closed 2019-06-05 by farrenv