ARMmbed/mbed-os-example-lorawan #147: Could not find executable for ARMC5

cipiasentini, 2019-06-19:
I'm using macOS, I've just installed Mbed cli (it supposedly contains GCC inside its package, nevertheless I downloaded from https://os.mbed.com/docs/mbed-os/v5.12/introduction/index.html and used the bin folder path for GCC_ARM_PATH variable). 
I've done everything from this guide but when trying to compile I get this error:
```
Could not find executable for ARMC5.
Currently set search path: /Users/user/Documents/MBED_LIBS/GCC/bin/
[mbed] ERROR: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python" returned error.
       Code: 2
       Path: "/Users/cip/Desktop/Grupo/mbed-os-example-lorawan"
       Command: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python -u /Users/cip/Desktop/Grupo/mbed-os-example-lorawan/mbed-os/tools/make.py -t ARM -m MTS_MDOT_F411RE --source . --build ./BUILD/MTS_MDOT_F411RE/ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

AnttiKauppila, 2019-06-19:
use GCC_ARM as your target. ARMC5 is commercial tool and if you have then check that your path has been configured properly.

cipiasentini, 2019-06-19:
> use GCC_ARM as your target. ARMC5 is commercial tool and if you have then check that your path has been configured properly.

Oh I see, now I finally got it to compile! Thanks! 

PS. You should consider adding this solution as a note on the README because I'm pretty sure that some other ARM newbie like me will face the same problem.

AnttiKauppila, 2019-06-19:
Great!

closed 2019-06-19 by AnttiKauppila