ARMmbed/mbed-os-example-lorawan #149: i2c communication example for MTS_MDOT_F411RE

cipiasentini, 2019-06-21:
I wanted to know if for the MTS_MDOT_F411RE module is there some example on how to use i2c bus to receive data from an Arduino. The only example I've seen online for i2c comm with mdots are from the official Multitech Dot-Examples from Mbed. But I never got it to work, nevertheless with this library the communication is working flawlessly.

janjongboom, 2019-06-22:
@cipiasentini The normal Mbed [I2C examples](https://os.mbed.com/docs/mbed-os/v5.12/apis/i2c.html) should work for that.

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila