ARMmbed/mbed-os-example-lorawan #163: Example broken for mbed-os master and 5.13.x

farrenv, 2019-09-10:
Target: EP_AGORA
The example works through mbed-os 5.12.4.  It breaks on master and all releases 5.13.x

On master I see:
[image]
5.13.4-5.13.2 Do not work at all
5.13.1 produces a hard fault
[image]
Call stack reveals LoRa thread call to be the culprit:
[image]

[image]

5.13.0 locks up in the same place as master:
[image]
Call stack reveals it is hanging out in sleep manager:
[image]

trowbridgec, 2019-09-10:
@ARMmbed/team-embeddedplanet @maclobdell

trowbridgec, 2019-09-10:
@bentcooke @AnttiKauppila

0xc0170, 2019-09-11:
Can you bisect, to find out what did break it? It would be good start.

AnttiKauppila, 2019-09-11:
It clearly runs out of memory, so something has been added into Mbed OS platform itself, because LoRaWAN stack is unmodified between those versions

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila

loverdeg-ep, 2019-09-13:
@AnttiKauppila @maclobdell @ranshe

This is an open source project. Internal to ARM there may be no resources to look at this, but that does not mean it is not still an active issue.

I understand why LoRaWAN is not currently a priority for ARM, but we still have paying customers using it and one of our top priorities is working towards enabling those customers to manage their devices via LWM2M; if you catch my drift.

0xc0170, 2019-09-16:
@loverdeg-ep Thanks for the feedback. What would you suggest? 

We shall not keep issues reported opened forever. However this issue is new so would expect to have it fixed within a month or become stale (might be closed later due to inactivity as "won't fix").

trowbridgec, 2019-09-27:
@maclobdell This is the issue I was trying to point you to. It's still a problem, so I'd argue it should be re-opened so ARM or someone else can address it.

farrenv, 2019-10-04:
@trowbridgec discovered a #define that limits any nordic target from using more than one Interruptin initialization with SDK 15.0 #11622