ARMmbed/mbed-os-example-lorawan #164: [Mbed OS OoB 5.14] System reboot due to fatal error

vince-zeng, 2019-09-11:
Compile the code for K64F platform as following:
`
mbed import mbed-os-example-lorawan
cd mbed-os-example-lorawan
mbed update mbed-os-5.14
mbed compile -m K64F -t GCC_ARM
`

Burn the code and get the following error output from UART:
```
++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20003408, Not allowed in ISR context
Location: 0xC939
Error Value: 0x20003408
Current Thread: rtx_idle  Id: 0x200021E8 Entry: 0xC945 StackSize: 0x200 StackMem: 0x20002530 SP: 0x2002FE48 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=K64F
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=1) reached maximum, system will halt after rebooting =

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20003408, Not allowed in ISR context
Location: 0xC939
Error Value: 0x20003408
Current Thread: rtx_idle  Id: 0x200021E8 Entry: 0xC945 StackSize: 0x200 StackMem: 0x20002530 SP: 0x2002FE48 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=K64F
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=2) reached maximum, system will halt after rebooting =
```

vince-zeng, 2019-09-11:
If compile with ARMC6, it can run well.  
But compile with IAR, 
```
mbed compile -m K64F -t IAR
```
get the similar error as GCC_ARM:
```
Error Message: Mutex: 0x20001B94, Not allowed in ISR context
Location: 0x0
Error Value: 0x20001B94
Current Thread: rtx_idle  Id: 0x20002558 Entry: 0x63C1 StackSize: 0x200 StackMem: 0x20001118 SP: 0x2002FE28 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=K64F
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=175) reached maximum, system will halt after rebooting =

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20001B94, Not allowed in ISR context
Location: 0x0
Error Value: 0x20001B94
Current Thread: rtx_idle  Id: 0x20002558 Entry: 0x63C1 StackSize: 0x200 StackMem: 0x20001118 SP: 0x2002FE28 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=K64F
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=176) reached maximum, system will halt after rebooting =
```

AnttiKauppila, 2019-09-11:
https://github.com/ARMmbed/mbed-os/pull/11412 Fixes this

closed 2019-09-11 by AnttiKauppila