ARMmbed/mbed-os-example-lorawan #194: Segmetation Fault Occurs on Downlink

factoryal, 2020-09-14:
Hi.

I'm trying to run mbed-os-example-lorawan with B-L072Z-LRWAN1 board.
only changes are some configurations on mbed_app.json

```json
"target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "lora.over-the-air-activation": true,
            "lora.duty-cycle-on": false,
            "target.components_add": ["SX1272", "SX1276", "SX126X"],
            "lora.phy": "KR920",
            "lora.device-eui": "(My 8-bytes device-eui key)",
            "lora.application-eui": "(My 8-bytes application-eui)",
            "lora.application-key": "(My 16-bytes application-key)"
        },
```

When downlink from TTN occurs, the whole program stops.
Running debug, I got this debug output.

[image]

```
Attached to debugger on port 50000
0003295:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 100480 bytes (785 pages) at 44.78 kB/s
Image loaded: \Users\vkdls\Mbed Programs\mbed-os-example-lorawan2\BUILD\DISCO_L072CZ_LRWAN1\ARMC6\mbed-os-example-lorawan2.elf
Note: automatically using hardware breakpoints for read-only addresses.
[New Thread 536878128]
[New Thread 536878196]
[New Thread 536878264]
[New Thread 536881424]
[Switching to Thread 536878128]
Thread 2 "main" hit Breakpoint 1, main () at ./main.cpp:96
96	    setup_trace();
[New Thread 2]
Thread 
6 "Handler mode" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 2]
0x080001c4 in HardFault_Handler ()
```

Is there any solution?
Or could I fix this problem?

Thanks.

SKRahulSK, 2020-09-25:
Hi @factoryal,

I am also facing the same problem when I send a downlink from TTN console the device.
Did you find any solution?

Thanks,
Rahul

factoryal, 2020-09-25:
@SKRahulSK Hi.
I didn't find any solution right now.

I tested for couple of weeks to find the reason why whole program stopped when downlink occures.
I finally run debug and found that segmentation fault was the reason. (I might find this earlier but not since pyOCD didn't worked for some reason 🤔)

I'm not a core developer. so I can't do any more at this moment.
I'm just waiting for their solution.
Hope they solve this problem quickly.

Thanks for reply. :)

PablOsma, 2021-01-29:
hi, 
try increasing the main thread stack size, that did the trick for me.

in mbed_app.json
[image]

of directly on mbed_config.h
[image]

good luck!

factoryal, 2021-02-19:
Hi. @PablOsma Thanks for reply.

I increased stack size of my board from 1024 to 2048 and it works!
I really appreciate for that. Thank you so much!

@SKRahulSK would you try this solution?

factoryal, 2021-02-19:
Plus, I also tried downlink from Chirpstack and it also works fine.