ARMmbed/mbed-os-example-lorawan #203: FaultType: HardFault

JchenEng, 2020-11-21:
### Description of defect
Hello,

I am still fairly new to the ARMmbed and lorawan world. When I finished adjusting the EUIs and build the program, I receive a FaultType: HardFault. 

I am using SX1276 as the lora-radio with loraphy being US915. 

Here is the serial communication messages that I receive from the terminal:

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000000
R1   : 20005000
R2   : 00000000
R3   : 00000000
R4   : 20003010
R5   : 20003E98
R6   : 2000208C
R7   : 200004BC
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 00000000
SP   : 20002058
LR   : 08011B03
PC   : 08005516
xPSR : 21000000
PSP  : 20002038
MSP  : 20004FC8
CPUID: 410CC601
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x8012C01
Error Value: 0x8005516
Current Thread: main  Id: 0x20002830 Entry: 0x8012E8D StackSize: 0x1000 StackMem                                     : 0x20001118 SP: 0x20004F70
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NUCLEO_L073R                                     Z
-- MbedOS Error Info --

#### Target(s) affected by this defect ?
Not really clear what targets are. 

#### Toolchain(s) (name and version) displaying this defect ?
I am using mbed studio as my toolchain. 

#### What version of Mbed-os are you using (tag or sha) ?
I am using mbed-os 5.14.0. 

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.3.1

#### How is this defect reproduced ?
Following the instructions.

JchenEng, 2020-11-23:
@ciarmcom 

Hi, I think the sha is #b6e5a0a8afa34dec9dae8963778aebce0c82a54b
[image]

pan-, 2020-11-23:
@JchenEng We do not support Mbed OS 5.14, have you tried with the latest version of Mbed OS (6.14) ? 
It would help if you provide more information about the application you try to execute and how to reproduce your issue.

JchenEng, 2020-11-23:
@pan-  I updated the mbed-os version to 5.15.6 sha: b114a9c878519d6489ac3426697196bbea34c8ea
[image]

However, I am encountering issues with the fsl_sha.h file, it says fsl_devie_registers.h file is not found and unknown type name: SHA_TYPE.

fsl_sha.h
mbed-os-example-mbed5-lorawan/mbed-os/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_MCU_LPC546XX/drivers
4
In included file: 'fsl_device_registers.h' file not foundclang(pp_file_not_found)[37, 10]
Unknown type name 'SHA_Type'clang(unknown_typename)[104, 19]
Unknown type name 'SHA_Type'clang(unknown_typename)[118, 21]
Unknown type name 'SHA_Type'clang(unknown_typename)[133, 21]