ARMmbed/mbed-os-example-lorawan #216: config/XX_example_config.json not in sync with current mbed_app.json

maxgerhardt, 2021-03-05:
### Description of defect

#### Target(s) affected by this defect ?

All
#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
PlatformIO (core 5.1.1.a)

#### How is this defect reproduced ?

Try and use this project with first with the `mbed_app.json` in the root of this repo. Everything works.

Now exchange the contents of the file with the content of e.g. `config/SX127X_example_config.json`. 

Compilation now gives

```
In file included from src\main.cpp:26:
src\lora_radio_helper.h:25:24: error: 'MBED_CONF_SX1272_LORA_DRIVER_SPI_MOSI' was not declared in this scope; did you mean 'MBED_CONF_SX126X_LORA_DRIVER_SLEEP_MODE'?
   25 | SX1272_LoRaRadio radio(MBED_CONF_SX1272_LORA_DRIVER_SPI_MOSI,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                        MBED_CONF_SX126X_LORA_DRIVER_SLEEP_MODE
src\lora_radio_helper.h:26:24: error: 'MBED_CONF_SX1272_LORA_DRIVER_SPI_MISO' was not declared in this scope; did you mean 'MBED_CONF_SX1272_LORA_DRIVER_BUFFER_SIZE'?
   26 |                        MBED_CONF_SX1272_LORA_DRIVER_SPI_MISO,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                        MBED_CONF_SX1272_LORA_DRIVER_BUFFER_SIZE
src\lora_radio_helper.h:27:24: error: 'MBED_CONF_SX1272_LORA_DRIVER_SPI_SCLK' was not declared in this scope; did you mean 'MBED_CONF_SX1272_LORA_DRIVER_BUFFER_SIZE'?
   27 |                        MBED_CONF_SX1272_LORA_DRIVER_SPI_SCLK,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                        MBED_CONF_SX1272_LORA_DRIVER_BUFFER_SIZE
src\lora_radio_helper.h:28:24: error: 'MBED_CONF_SX1272_LORA_DRIVER_SPI_CS' was not declared in this scope; did you mean 'MBED_CONF_SX1272_LORA_DRIVER_SPI_FREQUENCY'?
   28 |                        MBED_CONF_SX1272_LORA_DRIVER_SPI_CS,
..
```
because the `mbed_app.json` defines the pin as 

https://github.com/ARMmbed/mbed-os-example-lorawan/blob/890a6a919fb84e86fac210fed4daed24d3ff465c/mbed_app.json#L19-L31

whereas the template does it as e.g.

https://github.com/ARMmbed/mbed-os-example-lorawan/blob/890a6a919fb84e86fac210fed4daed24d3ff465c/config/SX126X_example_config.json#L37-L50

(`sx1276-lora-driver.spi-mosi` vs `lora-spi-mosi`  etc). 

It seems like the `mbed_app.json` was updated about a month ago but the templates are from november last year and don't represent the needed syntax anymore to describe the pins with the current driver version.

paul-szczepanek-arm, 2021-03-05:
Thanks for reporting this, looks like the the config folder is partially redundant and partially incorrect as you pointer out. I'll see if I can streamline the configs so we don't need the templates (or at least update the files).

jeromecoutant, 2021-03-08:
See https://github.com/ARMmbed/mbed-os-example-lorawan/pull/206

paul-szczepanek-arm, 2021-03-19:
https://github.com/ARMmbed/mbed-os-example-lorawan/pull/219

paul-szczepanek-arm, 2021-03-22:
Fix merged

closed 2021-03-22 by paul-szczepanek-arm