ARMmbed/mbed-os-example-lorawan #217: send/receive data to and from another lora end node while sending data to network server

sarwadenj, 2021-03-18:
### Description of defect--> 

I'm using mbed-os-example-lorawan to send data to assigned network server. However I want my lora nodes to send and receive DATA  from another lora end node. How can I configure that. Please guide.

Board: Disco-lrwan1(BL072-LRWAN1)

maxgerhardt, 2021-03-22:
*LoRaWAN* inherently has a structure with nodes and gateways. Gateways send their transmission (downlink) with the "IQ polarity inversion" option (see [here](https://www.thethingsnetwork.org/forum/t/what-prevents-gateways-from-hearing-eachother/19278/3)) so that only nodes hear them, which also listen with IQ inversion option, and not other gateways. A node's transmission is however heard by every gateway, since no IQ inversion is done and the radio filters for its own device address. 

So with that in place, the only way to achieve a LoRaWAN between two nodes is to modify one of the drivers to act like a gateway, by using the same IQ polarity in listening as the other node uses for sending. But there are other hacks to be done regarding using the device address of your target listener and the content of the header in the LoRaWAN message (there's a message type in there). So I'd totally not recommend this. 

Usually when people want to node-to-node communication they use raw LoRa packets. You might e.g. want to try out https://github.com/sandeepmistry/arduino-LoRa with the `LoRaReceiver` and ` LoRaSender` examples.

sarwadenj, 2021-03-22:
@maxgerhardt Thanks! 

Is there any way, where end node can send and receive data from end node as well as gateway in one application?