ARMmbed/mbed-os-example-lorawan #223: Integrate LR1110 and mbed-os

kvkc97, 2021-04-26:
I want t use LR1110 as the LoRa driver for the lorawan example code. I have added the LR1110 driver in mbed-os\connectivity\drivers\lora folder. A pull request for the same was added to the mbed-os and link is given below. 
https://github.com/ARMmbed/mbed-os/pull/14560
I am using MBED CLI2 to compile and added STM32WB55RGV6 as a custom target (Custom target addition and compilation was successful). I modified the lora_radio_helper.h file to include the lr1110 driver. The file is attached.
[lora_radio_helper.txt](https://github.com/ARMmbed/mbed-os-example-lorawan/files/6375212/lora_radio_helper.txt)

I am getting compilation error as attached.
[image]

0xc0170, 2021-04-26:
First, the config should be all uppercase (lr1110). Can you check if mbed_config.h contians the definitions that are missing?

0xc0170, 2021-04-26:
If you look at drivers, for instance GEMALTO, you can find mbed_lib.json config where pins are defined

```
{
    "name": "GEMALTO_CINTERION",
    "config": {
        "tx": {
            "help": "TX pin for serial connection. D1 assumed if Arduino Form Factor, needs to be set/overwritten otherwise.",
            "value": null
        },
        "rx": {
            "help": "RX pin for serial connection. D0 assumed if Arduino Form Factor, needs to be set/overwritten otherwise.",
            "value": null
        },
        "rts": {
            "help": "RTS pin for serial connection",
            "value": null
        },
        "cts": {
            "help": "CTS pin for serial connection",
            "value": null
        },
        "baudrate" : {
            "help": "Serial connection baud rate",
            "value": 115200
        },
        "provide-default": {
            "help": "Provide as default CellularDevice [true/false]",
            "value": false
        }
    }
}

```

I assume you need something similar for your radio driver

kvkc97, 2021-04-26:
I do not have mbed_config.h file in my application. Do I need to create one?
Also I can find a mbed_lib.json file in mbed-os\drivers folder but not in the connectivity folder. Is this file to be created?

0xc0170, 2021-04-26:
If your app is using the config values, they need to come from somewhere. You should add mbed_lib.json to your driver you are adding. The default value I assume would be null. An application, would overwrite this to specific layout you got on your board. 

This should fix the issue.

kvkc97, 2021-05-04:
I have added the mbed_lib.json file but still could not understand how and where to define the MBED_CONF_LR1110 variables. I have attached the mbed_lib.json file.

[image]

jeromecoutant, 2021-05-05:
Check cmake_build/<TARGET>/develop/GCC_ARM/mbed-os/compile_time_defs.txt

kvkc97, 2021-05-05:
I checked the above mentioned file and found the variables but could not understand how to move forward with it. Also in the mbed_lib.json file, I have added value NC for spi-mosi, spi-miso and spi-sclk and an error is generated in command prompt. I have attached the error and also the mbed-lib.json file and the compile_time_defs.txt file.
[compile_time_defs.txt](https://github.com/ARMmbed/mbed-os-example-lorawan/files/6428890/compile_time_defs.txt)

mbed-lib.json file
[image]

Compiler error
[image]

jeromecoutant, 2021-05-06:
Defintion looks OK in mbed_lib.json.
Then you are supposed to define them in your local mbed_app.json
  "lr1110-lora-driver.spi-mosi":"XXX"

kvkc97, 2021-05-06:
I updated the mbed_app.json file with pin names as per the mentioned format and the following error is obtained.
[image]

kvkc97, 2021-05-15:
Can I have an update regarding this issue.
@jeromecoutant @0xc0170 @matthieuantoine

0xc0170, 2021-05-17:
Where are your pinnames defined? THe error indicates the compiler can't resolve `PB_15` for instance. It should come from your PinNames.h. 

Regarding the custom target,is it linked properly? There was a recent discussion about it , see https://github.com/ARMmbed/mbed-tools/issues/278#issuecomment-838255135 if it helps

kvkc97, 2021-05-17:
The pinnames are defined in PinNames.h file in the custom target's folder. Have attached the file. The same pins can be defined successfully for SX1272 Radio.
 
[PinNames.txt](https://github.com/ARMmbed/mbed-os-example-lorawan/files/6492737/PinNames.txt)

I believe the custom target addition is successful as the program is compiled successfully when SX1272 is selected as LoRa radio.

0xc0170, 2021-05-18:
it would be helpful to provide an example that would reproduce an issue (possibly with regular target, adding this new lr1110 driver and get this error).

kvkc97, 2021-05-18:
I tested with K64F target which is a predefined board in MBED CLI2. Default lora driver is SX1276 whose compilation is successful. However, on adding the custom lora driver LR1110, configuration error is occurred. To add LR1110, I modified the mbed_app.json file in the base directory as below. Error is also attached.
[image]

[image]

kvkc97, 2021-05-26:
Can I have an update on this query

0xc0170, 2021-05-31:
The latest errors might be better to be reported to mbed-tools. It is not obvious what element was not found. What is printed if you use `-v` - verbose mode ?

kvkc97, 2021-05-31:
The latest errors are cleared but the pin definition error is still there. I tried to compile using three target board and all three of them are giving the same result, ie, the code is compiled for driver SX1276 but not for LR1110 (custom driver). The same pin definition error as previously attached is displayed.
The three target boards I used include 2 regular targets (K64F and NUCLEO_WB55RG) and 1 custom target (created by me for STM32WB55RGV6)

0xc0170, 2021-06-01:
Can you create a simple code snippet or provide a branch we can use to reproduce the issue locally?

kvkc97, 2021-06-01:
I have created a pull request of my code #228

kvkc97, 2021-06-06:
Hi,

I modified the SX1276 radio files(from COMPONENT_SX1276 folder in mbed-os) with LR1110 radio functions. The code is compiled properly. However, I cannot change the generic pin names(MBED_CONF_SX1276_LORA_DRIVER_NAME) given in the radio files. Is there a way to solve this and also the previous issue of adding custom driver.

kvkc97, 2021-06-16:
Can I have an update on this case

kvkc97, 2021-06-28:
Is there any update on this case.

Thank you

paul-szczepanek-arm, 2021-06-28:
The PR is too big, please just include the changes needed - rebase onto the branch you're making the PR against.

Are you sure you have the configuration options defined in your new driver? Looks like the configuration tools can't find the options you are trying to override in your mbed_app.json.

kvkc97, 2021-06-29:
I have my PR with latest files. The error is persisting. I have mentioned the configuration options I did in my PR. However, I am not sure if there is something missing. The PR is #234