ARMmbed/mbed-os-example-lorawan #227: Can not reach low power state when radio is initialized outside of main()

charlesmodrich-tc, 2021-04-30:
### Description of defect

Inability to enter sleep mode (or at least reach expected sleep mode average current) after calling 
`static LoRaWANInterface lorawan(radio);` when working with NRF52840-DK + SX1262MB2xAS mbed shield

even when building with Develop profile. Results in average "sleep" current of ~6.5mA. Uncommenting all code before main() (except for above line) and removing all code within main() yields expected sleep current (~14uA).

#### Target(s) affected by this defect ?
NRF52840-DK

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio 1.4.0

#### What version of Mbed-os are you using (tag or sha) ?
6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed CLI 2

#### How is this defect reproduced ?
Set up 2 NRF52840 DK's: one with PowerProfilerKit, another with SX1262MB2xAS mbed shield. Connect both to PC via microUSB (not nRF USB). Have PPK DK power LoRa shield DK via External supply, switch power on LoRaDK ON, nRF ONLY switch, VEXT->nRF switch to ON. Load up nRF Connect utility for the power profiler kit application, and enable DUT power and monitoring. 

Use the default example for this sketch, but remove most of the code such that all that remains looks as below:

`#include <stdio.h>

#include "lorawan/LoRaWANInterface.h"
#include "lorawan/system/lorawan_data_structures.h"
#include "events/EventQueue.h"

// Application helpers
#include "SX126X_LoRaRadio.h"
SX126X_LoRaRadio radio(D11,//MOSI
                       D12,//MISO
                       D13,//SCK
                       D7,//CS/NSS
                       A0,//RESET
                       D5,//DIO1
                       D3,//BUSY
                       A1,//FREQ_SEL
                       A2,//DEV_SEL
                       A3,//XTAL_SEL
                       D8);//ANT_SW

#include "mbed.h"

//DigitalOut led1(LED1);
using namespace events;
uint8_t tx_buffer[30];
uint8_t rx_buffer[30];
#define TX_TIMER                        20000
#define MAX_NUMBER_OF_EVENTS            10
#define CONFIRMED_MSG_RETRY_COUNTER     3
#define PC_9                            0

static EventQueue ev_queue(MAX_NUMBER_OF_EVENTS *EVENTS_EVENT_SIZE);
static void lora_event_handler(lorawan_event_t event);
static lorawan_app_callbacks_t callbacks;

static LoRaWANInterface lorawan(radio);

int main()
{       
rtos::ThisThread::sleep_for(1000);
}`

change the following in mbed_app.json:

`"target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "lora.over-the-air-activation": true,
            "lora.duty-cycle-on": false,
            "lora.phy": "US915",
            "lora.device-eui": "{ 0x00, 0x80, 0xA0, 0xD0, 0x09, 0xEA, 0x82, 0x37 }",
            "lora.application-eui": "{ 0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x03, 0xE4, 0x02 }",
            "lora.application-key": "{ 0xEE, 0xB6, 0x72, 0x7D, 0xEB, 0x52, 0x08, 0x1C, 0x2E, 0x32, 0xC4, 0x3B, 0xB8, 0x96, 0x86, 0x6D }",
            "target.components_add":            ["SX126X"],
            "lora.app-port": 2,
            "lora.fsb-mask": "{0xFF00, 0x0000, 0x0000, 0x0000, 0x0002}"
        },`

`"macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_lora_config.h\"", "MBED_TICKLESS=1"]`

Now, compile, upload and compare the average currents when `static LoRaWANInterface lorawan(radio);` is commented, and uncommented. For me, results for commented are ~14uA average sleep current over 52ms period, while commented shows an average sleep current of approximately 6.5mA over 52ms period.

evandavey, 2021-06-26:
@charlesmodrich-tc  I have similar issues in #222.  Did you find a solution?  For tcxo control, I am exploring modifying the set_operation_mode call in the driver i.e. set to 0 when RF_OPMODE_SLEEP otherwise 1.  But that still leaves I high residual.

charlesmodrich-tc, 2021-06-28:
Unfortunately no, I wasn't able to find a solution. Ultimately I was forced
to move away from Mbed OS for this reason since I wasn't able to get the
low current performance all of my devices would require.

evandavey, 2021-06-28:
@charlesmodrich-tc  - very frustrating.

charlesmodrich-tc, 2021-06-28:
It really is! I spent a lot of time and hope on this before giving up.

evandavey, 2021-06-28:
Yes, have been in the same boat!  Having to revert back to the manufacturer SDKs is its own path of pain given all the messy #defines to support multiple targets and tight coupling of business logic to SDK logic.  It's a shame support resources for MBED seem to have dropped off recently and it is stuck being half a corporate project and half an open source project with neither seeming to have particularly strong momentum.

evandavey, 2021-07-22:
@charlesmodrich-tc - I have been doing some more debugging, this time on a Murata ISJ dev board ([link](https://www.digikey.com.au/products/en/rf-if-and-rfid/rf-evaluation-and-development-kits-boards/859?k=murata%201sj)).  This is a STM32L072 + SX126X radio.  With the Murata supplied AT firmware, I measure this board at the expected 2uA, so I know this is the baseline.   I am using a new example from scratch, but using the master branch of mbed-os (as I needed to make some non power related changes to the SX126X driver).

With the code below, I am able to achieve an average of 50uA.  Interestingly, if I power up the board (which puts the radio to sleep), then re-upload the code with the radio.sleep() line commented out, current falls to 11uA, which is very similar to the 14uA you measured.  I think this is related to GPIO/peripheral configuration in sleep but I haven't been able to figure out why, or where I am losing 9uA relative to the AT firmware.

mbed_app.json
```
{
  "requires": [
    "bare-metal",
    "events",
    "storage",
    "spif-driver",
    "lora",
    "mbedtls",
    "sx1276-lora-driver",
    "SX126X-lora-driver"
  ],
  "target_overrides": {
    "*": {
      "target.macros_add": ["MBED_TICKLESS"],
      "target.lpticker_lptim": 0
    },
    "MURATA_TYPE1SJ_EVB": {
      "SX126X-lora-driver.freq-support": "MATCHING_FREQ_915",
      "SX126X-lora-driver.device-variant": "SX1262"
    }
  }
}
```
main.cpp
```
#include "mbed.h"
#include "events/EventQueue.h"
#include "radio.h"
#include "SX126X_LoRaRadio.h"

#define MAX_NUMBER_OF_EVENTS 20

static EventQueue ev_queue(MAX_NUMBER_OF_EVENTS *EVENTS_EVENT_SIZE);

SX126X_LoRaRadio radio(LORA_SPI_MOSI,
                       LORA_SPI_MISO,
                       LORA_SPI_SCLK,
                       LORA_SPI_CS,
                       LORA_RESET,
                       LORA_DIO1,
                       LORA_BUSY,
                       LORA_FREQ_SELECT,
                       LORA_DEVICE_SELECT,
                       LORA_CRYSTAL_SELECT,
                       LORA_ANT_SWITCH);

int main(void)
{

    radio.sleep();
    ev_queue.dispatch_forever();
}
```

jeromecoutant, 2021-07-22:
>       "target.lpticker_lptim": 0

This is not tested, not validated, so not recommended...
Keep the default configuration

evandavey, 2021-07-22:
@jeromecoutant - this line saves about 30uA .  At the moment my objective is to just get current down to 2uA with nothing running, then start adding back functionality to be able to do a full LoRaWAN test so I'll admit I've been trying things such as that line without fulling understanding the impact.  The 2uA firmware is written in https://www.st.com/en/embedded-software/i-cube-lrwan.html but I find the structure such a headache to deal with.

hallard, 2021-07-22:
That's interesting because I wanted to post a comment this night but went to bed before. In the meantime conversation has new comments.
I'm currently switching to mbed for my lorawan E5 and RAK3172 board because it's really nice simple with lot of features and this lorawan example works.
But my next step is to go to low power testing and seing this makes me feel inconfortable because the goal of LoRaWAN device is to be low lower. I'm going to 1.5uA with old murata one on custom code without mbed so 11uA or 50uA is not an option on my side.
Would be interesting to have @jeromecoutant, or @MarceloSalazar on this point, in the meantime I will give some tries on my boards.

hallard, 2021-07-22:
> The 2uA firmware is written in https://www.st.com/en/embedded-software/i-cube-lrwan.html but I find the structure such a headache to deal with.

I really don't understand how can this structure give you headache (kidding), but I'm glad I don't feel alone on this one 👍

evandavey, 2021-07-22:
@hallard  - I have some E5's on order along with the https://www.seeedstudio.com/LoRa-E5-mini-STM32WLE5JC-p-4869.html.   Is there a target already that works with this chip and can give low power?  I don't care what chip I use at this point as long as I can get something functional - don't know if there is anyone using MBED lorawan in the real world.

As to i-cube-lrwan, honestly my adaptions are relatively minor and I could probably mod it to get it working, but :scream: I don't the business logic would be so tightly coupled with that structure I couldn't deliver in good conscience.

hallard, 2021-07-22:
> @hallard - I have some E5's on order along with the https://www.seeedstudio.com/LoRa-E5-mini-STM32WLE5JC-p-4869.html. Is there a target already that works with this chip and can give low power? 

Sure take a look [there](https://github.com/ARMmbed/stm32customtargets#seeed-studio-lora-e5) (not tested on Low Power yet )

for STM32WL i-cude-lrwan has been replaced by [STM32Cube MCU Package for STM32WL series(SDK)](https://my.st.com/content/my_st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-mcu-mpu-packages/stm32cubewl.license=1608693595598.product=STM32CubeWL.version=1.0.0.html#overview) may be it was too simple.
More information [here](https://wiki.seeedstudio.com/LoRa_E5_Dev_Board/) but don't dream, file and folder architecture won't remove your any headache

evandavey, 2021-07-22:
@hallard thanks for the links.  Not Low Power tested is the concern.  I've tried the Type ABZ (custom build and DISCO LRWAN1), Type 1SJ and STM32L0/RFM95 (draguino LNS50 node) so far with no luck for easy low power < 10uA (and even struggling for <100uA).  I guess the trade-off for using a simplified framework like MBED is you lose that fine-grain control to library implementations.  Still, it such a common problem I'd hope there would be sufficient community support / documentation to get things running at least with a viable battery life for a low power long-term node (i.e. 1 year plus).  Especially for dev boards like the DISCO LRWAN1.

hallard, 2021-07-22:
Worth looking at this sample code https://os.mbed.com/users/kenjiArai/code/Check_DeepSleep_os6/
For sure need to be adapted for STM32 target but could be an excellent starting point because on some board got 2uA which is both our target :-)

evandavey, 2021-07-22:
@hallard I have attempted something like the "LowPowerConfiguration" function before.  Firstly, I don't think "AHBENR" exists for a L0 target (so I switched for "IOPENR").  Secondly, I think the presence of the radio messes with this code as I have experienced worse performance after calling such a function following radio.sleep (as in mA).    Otherwise, I don't see anything obvious in this code that is doing anything I haven't tried before.  I haven't found anything documented as to what happens to peripherals/gpio when deep sleep is called.

evandavey, 2021-07-22:
BTW, with the exact some code and the LNS50 as a target (so swapping the SX126X driver for SX1276) I see 500uA.  But I am not convinced the pin mappings for the SX1276 are correct for the RFM95 module (https://github.com/dragino/Lora/blob/master/LSN50/v2.0/LoRa%20ST%20Sensor%20Node%20v2.0.sch.pdf and https://raw.githubusercontent.com/dragino/Lora/master/LoRaST/v1.0/LoRa%20ST%20v1.0_Sch.pdf).

hallard, 2021-07-22:
Ok done some testing on my LoRa-E5 board, after removing all code of the sample (and all GPIO config such as LED and Button) here we go

[image]

1.43uA in deep sleep mode, perfect !!

`mbed_app.json`
```json
{
    "requires": ["bare-metal", "events", "lora", "mbedtls"],
    "target_overrides": {
        "*": {
            "target.printf_lib": "std"
        }
    }
}

```

code
```cpp
//  Include --------------------------------------------------------------------
#include "mbed.h"

//  Constructor ----------------------------------------------------------------
//DigitalIn   my_sw(BUTTON1);
//DigitalOut  myled(LED1,1);
static BufferedSerial pc(CONSOLE_TX, CONSOLE_RX, 115200);

//------------------------------------------------------------------------------
//  Control Program
//------------------------------------------------------------------------------
int main()
{
    while (true) {
//        myled = 0 ; // ON
//        ThisThread::sleep_for(1s);
//        myled = 1 ; // Off
//        ThisThread::sleep_for(2s);
        hal_deepsleep();
    }
}

```
Please not that using `ThisThread::sleep_for(2s)` instead sleep at 3mA

evandavey, 2021-07-22:
@hallard - great, sounds promising.  Is this the Seeed Studio dev board or one of your own?  I think with bare metal you need to use thread_sleep_for rather than ThisThread::sleep_for for the sleep manager to work (or events.dispatch_forever() like I used).  What's happening with the radio?  Is it defaulting to off for this board?  What are the settings for PeripheralPins.c?  What is the system clock configuration?

hallard, 2021-07-23:
@evandavey yes it's my basic own [board](https://github.com/hallard/LoRa-E5-Breakout), pin definitions are [here](https://github.com/ARMmbed/stm32customtargets/tree/master/TARGET_STM32WL/TARGET_LORA_E5) and the radio is included into the STM32WL chip.

`hal_deepsleep()` may be not the best because I'm not sure we can wake up with a timer, but I'm wondering how the difference between with `thread_sleep_for()`

As you can see as soon as I'm using this code consumption in sleep goes from 1.5uA to 2.9mA

```cpp
int main()
{
    while (true) {
          thread_sleep_for(5000);
    }
}
```

[image]

evandavey, 2021-07-23:
@hallard  - I might be wrong on thread_sleep_for https://forums.mbed.com/t/clarification-on-os-6-empty-vs-os-6-bare-metal-example-programs-1-0-0/8950 and it is equivalent and doesn't call the sleep manager.  What do you get using events.dispatch_forever()?  Maybe the radio defaults to its sleep state on the STM32WL?

hallard, 2021-07-23:
Same thing with
```cpp
int main()
{
    EventQueue queue;
    queue.dispatch_forever();
}
```

2.9mA

evandavey, 2021-07-23:
Same code but not using bare-metal?

hallard, 2021-07-23:
Sorry same code, same config so yes `bare-metal`

evandavey, 2021-07-23:
Sorry, I meant, what do you get running the same code but not in bare-metal?

hallard, 2021-07-23:
```cpp
int main()
{
    EventQueue queue;
    queue.dispatch_forever();
}
```

`mbed_app.json` => 6.5mA
```json
{
    "requires": ["bare-metal", "events", "lora", "mbedtls"],
    "target_overrides": {
        "*": {
            "target.printf_lib": "std",
            "target.macros_add": ["MBED_TICKLESS"],
            "target.lpticker_lptim": 0
        }
    }
}
```

`mbed_app.json` => 2.9mA
```json
{
    "requires": ["bare-metal", "events", "lora", "mbedtls"],
    "target_overrides": {
        "*": {
            "target.printf_lib": "std",
            "target.macros_add": ["MBED_TICKLESS"]
        }
    }
}
```

evandavey, 2021-07-23:
So  "target.lpticker_lptim": 0 saves 3.6mA?  

For not bare-metal I mean

```c++
int main()
{
    EventQueue queue;
    queue.dispatch_forever();
}
```

```json
{
    "target_overrides": {
        "*": {
            "target.printf_lib": "std"
        }
    }
}
```

hallard, 2021-07-23:
and here we go without `bare-metal`, same results

```cpp
int main()
{
    EventQueue queue;
    queue.dispatch_forever();
}
```

`mbed_app.json` => 6.5mA
```json
{
    "target_overrides": {
        "*": {
            "target.printf_lib": "std",
            "target.macros_add": ["MBED_TICKLESS"],
            "target.lpticker_lptim": 0
        }
    }
}
```

`mbed_app.json` => 2.9mA
```json
{
    "target_overrides": {
        "*": {
            "target.printf_lib": "std",
            "target.macros_add": ["MBED_TICKLESS"]
        }
    }
}
```

`mbed_app.json` => 2.9mA
```json
{
    "target_overrides": {
        "*": {
            "target.printf_lib": "std"
        }
    }
}
```

evandavey, 2021-07-23:
I've done a little more testing myself.

3.4ma
```c++
int main()
{
    EventQueue queue;
    queue.dispatch_forever();
}
```
2.4ma
```c++
#include "SX126X_LoRaRadio.h"
SX126X_LoRaRadio radio(LORA_SPI_MOSI,
                       LORA_SPI_MISO,
                       LORA_SPI_SCLK,
                       LORA_SPI_CS,
                       LORA_RESET,
                       LORA_DIO1,
                       LORA_BUSY,
                       LORA_FREQ_SELECT,
                       LORA_DEVICE_SELECT,
                       LORA_CRYSTAL_SELECT,
                       LORA_ANT_SWITCH);

int main()
{
    EventQueue queue;
    queue.dispatch_forever();
}
```

70uA
```c++
#include "SX126X_LoRaRadio.h"
SX126X_LoRaRadio radio(LORA_SPI_MOSI,
                       LORA_SPI_MISO,
                       LORA_SPI_SCLK,
                       LORA_SPI_CS,
                       LORA_RESET,
                       LORA_DIO1,
                       LORA_BUSY,
                       LORA_FREQ_SELECT,
                       LORA_DEVICE_SELECT,
                       LORA_CRYSTAL_SELECT,
                       LORA_ANT_SWITCH);

int main()
{
    EventQueue queue;
    radio.sleep();
    queue.dispatch_forever();
}
```

```json
{
  "requires": [
    "bare-metal",
    "events",
    "lora",
    "mbedtls",
    "SX126X-lora-driver"
  ],
  "target_overrides": {
    "*": {
      "target.macros_add": ["MBED_TICKLESS"],
      "target.reset_gpio_on_init": 1
    },
    "MURATA_TYPE1SJ_EVB": {
      "SX126X-lora-driver.freq-support": "MATCHING_FREQ_915",
      "SX126X-lora-driver.device-variant": "SX1262"
    }
  }
}

```

evandavey, 2021-07-23:
I am pretty sure that in both bare-metal and RTOS queue->dispatch_forever() should go into deep sleep when idle (and this is what I am experiencing).  Can you turn on sleep tracing to see if anything is blocking deep sleep?

hallard, 2021-07-23:
May be the Serial port console? do you know how to be sure that any serial is disabled?

evandavey, 2021-07-23:
You could try: mbed_file_handle(STDIN_FILENO)->enable_input(false)
[Console and Deep Sleep](https://os.mbed.com/docs/mbed-os/v6.12/apis/power-management-sleep.html)

hallard, 2021-07-23:
With radio driver

```cpp
#include "mbed.h"
#include "STM32WL_LoRaRadio.h"

static BufferedSerial pc(CONSOLE_TX, CONSOLE_RX, 115200);
STM32WL_LoRaRadio radio;

int main()
{
    EventQueue queue;
    radio.sleep();
    queue.dispatch_forever();
}
```

`mbed_app.json` => 6mA average
```json
{
    "requires": ["bare-metal", "events", "lora", "mbedtls", "stm32wl-lora-driver"],
    "target_overrides": {
        "*": {
            "target.printf_lib": "std"
        }
    }
}
```

[image]

evandavey, 2021-07-23:
And you're using a develop or release build profile (not debug?)

hallard, 2021-07-23:
ah ah good catch, it's a debug, compiling above code as release and here we go

Avg 1.7uA and peak are 4mA for approx 2ms

[image]

evandavey, 2021-07-23:
OK, that's great!! And is it the same without the radio.sleep() call?

Perhaps we need to compare the differences between "STM32WL_LoRaRadio.h" and "SX126X_LoRaRadio.h".

hallard, 2021-07-23:
yes same things without `radio.sleep()`, I already had 1.5uA with murata ABZ (STM32L0 + SX126x) with another framework so it's do able and may be your issue is related to mbed. Do you have same issue on mbed 5 where ABZ is officially supported?

evandavey, 2021-07-23:
What framework?  I have tried stm32duino / arduino-lmic but similar issues.  I am actually using a Type 1SJ for these tests.  My LRWAN1 board had to be repurposed in a field test and I haven't re-checked on my custom board.  But experiencing similar issues on the Draguino LNS50.  Definetely there is some config issues with the framework, but they should be solveable with support....

Please post your results when you have done a end-to-end test with LoRaWAN active.

hallard, 2021-07-23:
using this one https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0
By the way do your board manage USB because I remember having issues with that, very tricky, needed to disable some USB stuff and more, needed to reset (start the board) without usb connected, removing USB cable once started prevented to go to low power.

evandavey, 2021-07-23:
I used that GrumpyOldPizza framework very early on but ran into LoRa implementation issues, can't remember exactly what now, and gave up.  Could have been related to early days of them supporting AU915.  If you have any code you can share that would be greatly appreciated.

I experience a similar issue with the Draguino LNS50 where I have to use a ST-Link and need it unplugged for low power.  Have tested the Type 1SJ both with and without the (on-board) debugger connected and it makes no difference (and the ~45uA I am seeking is too low for that anyway).

hallard, 2021-07-23:
Here part of my code working for GrumpyOldPizza (with git version since some API have changed since)

```cpp

#include "STM32L0.h"
#include "LoRaWAN.h"
#include "TimerMillis.h"

// Transmit period in s
#define LORA_TRANSMIT_PERIOD  300

// for USB
#define DEBUG_USB
#define DEBUG_SERIAL 

// Disable USB after boot
#define DISABLE_USB_SERIAL

#ifdef  DEBUG_SERIAL
#define SERIAL_DEBUG  Serial1
#elif defined (DEBUG_USB)
#define SERIAL_DEBUG  Serial
#endif

#if defined (DEBUG_USB) || defined (DEBUG_SERIAL)
#define debug(VAL)          { SERIAL_DEBUG.print(VAL); }
#define debug2(VAL, MODE)   { SERIAL_DEBUG.print(VAL, MODE); }
#define debugln(VAL)        { SERIAL_DEBUG.println(VAL); }
#define debug2ln(VAL, MODE) { SERIAL_DEBUG.println(VAL, MODE); }
#define debugflush()        { SERIAL_DEBUG.flush(); }
#else
#define debug(VAL)          {}
#define debug2(VAL, MODE)   {}
#define debugln(VAL)        {}
#define debug2ln(VAL, MODE) {}
#define debugflush()        {}
#endif

#define WAKE_BTN YOUR_BUTTON_IO_NAME

uint32_t transmitPeriod = LORA_TRANSMIT_PERIOD; 
bool VUSB, hasUSB;

volatile bool wake_btn = false; 
volatile bool wake_timer = false; 
volatile bool wake_join = false; 

TimerMillis wakeTimer;

void ledOff()
{ 
  // Letting PWM active prevent stop mode and thus
  // stop()/deepsleep() does not go into sleep mode.
  pinMode(LED_RED, OUTPUT);     
  pinMode(LED_GRN, OUTPUT);     
  pinMode(LED_BLU, OUTPUT); 
  digitalWrite(LED_RED, LEDOFF);     
  digitalWrite(LED_GRN, LEDOFF);     
  digitalWrite(LED_BLU, LEDOFF);     
}

void btn_interrupt()
{
  wake_btn = true; 
  STM32L0.wakeup();
}

void timer_interrupt(void)
{
  wake_timer = true;
  STM32L0.wakeup();
}

void joinCallback(void)
{
  debug("joinCallback() ");
  wake_join = true;
  STM32L0.wakeup();
}

void setup() 
{
  char buffer[18];

  VUSB = USBDevice.attached();
  hasUSB = USBDevice.connected();

  pinMode(WAKE_BTN, INPUT);

  LoRaWAN.begin(EU868);
  LoRaWAN.getDevEui(buffer,18);

#ifdef DEBUG_SERIAL
  Serial1.begin(115200);
  while (!Serial1 && millis()<=2000 ) {
  }
  Serial1.println("Serial1 OK");
#endif

#ifdef DEBUG_USB
  Serial.begin(115200);

  while (!Serial && millis()<=5000 ) {
  }
  Serial.println("OK");
#endif

  LoRaWAN.setADR(true);
  LoRaWAN.setDutyCycle(false);
  LoRaWAN.onJoin(joinCallback);
  LoRaWAN.onReceive(receiveCallback);

  debugln("LoRa Joining");  
  LoRaWAN.joinOTAA(appEui, appKey, (const char *) buffer);

#ifdef DEBUG_USB
#ifdef DISABLE_USB_SERIAL
  // Switch to real uart
  Serial.println("Stopping USB");
  Serial.end();
#else
  Serial.println("!!!! Leaving USB Serial running !!!!");
#endif
#endif

// In all case even just Serial 
#ifdef DISABLE_USB_SERIAL
  USBDevice.detach();
#endif

  // Initialize our wake timer but do net setup yet
  wakeTimer.start( timer_interrupt, 10000);
  wakeTimer.stop();
}

void loop() 
{
  unsigned long next_transmit = transmitPeriod;

  if (wake_btn) {
    debugln("button");  
  }

  if (wake_join) {

    if (!LoRaWAN.joined()) {
      debug("NOT ");
    }
    debugln("JOINED");
    wake_join = false;
  }

  if (wake_timer) {
    // Security to be sure to be waked
    wake_timer = false;
  }

  ledOff();
  wake_btn = false;

  // reset heartbeat every 4 x transmit period
  wakeTimer.restart( LORA_TRANSMIT_PERIOD * 4 * 1000 + 5000);

	debug("Going to Sleep ("); 
  debug(next_transmit); 
  debugln("s)..."); 
  debugflush();

  attachInterrupt(WAKE_BTN, btn_interrupt, RISING);   

  STM32L0.deepsleep(next_transmit*1000);

   // Just waked up
  detachInterrupt(WAKE_BTN);   
}
```

evandavey, 2021-07-23:
Awesome, thanks!  Can't see too much different from MBED in this application code so it must be the initial pin config / deep sleep code that is done differently.

hallard, 2021-07-23:
can you share your HW schematics or not?

evandavey, 2021-07-23:
Type 1SJ EVM I think technically can't be shared (though they give you access if you buy the board) - I asked if I can share my target files with the mbed-os community but so far no response.  My custom board I can't as it is a client design (but I have accounted for TCXO) - it is pretty much the reference design with provision for an external crystal and for TCXO to be GPIO switched.  I'd build another one to test with bare minimum components but can't source the Type ABZ modules at the moment.  This issue shows up on the DISCO LRWAN1 board too.  If I dig in to the SX1276 driver though you supply a TCXO pin I don't think it actually does anything with it so I have manually disabled it.  You can measure the impact (about 1mA from memory) but that doesn't help (see https://github.com/ARMmbed/mbed-os-example-lorawan/issues/222).

hallard, 2021-07-23:
Here mine

D12 is PB14 in my case 

[image]

evandavey, 2021-07-23:
Does the "LoRaWAN.h" switch it?  If I search _tcxo in mbed SX1276_LoRaRadio.cpp I can see it turned on on line 212 but then not referenced again.

evandavey, 2021-07-23:
Yep  - I can see specific calls in the libraries and looks like the GPIOs are configured better too when it's off for deep sleep:
```c++
# https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/b1cf1cd6eba2f06bbe4edbe76aa43ae5971e8d8c/system/STM32L0xx/Source/LoRa/Boards/cmwx1zzabz-board.c
# Line 134
void SX1276SetBoardTcxo( bool state )
{
    if (RADIO_TCXO_VCC != STM32L0_GPIO_PIN_NONE)
    {
        if( state == true )
        {
            stm32l0_gpio_pin_configure(RADIO_TCXO_VCC, (STM32L0_GPIO_PARK_NONE | STM32L0_GPIO_PUPD_NONE | STM32L0_GPIO_OSPEED_HIGH | STM32L0_GPIO_OTYPE_PUSHPULL | STM32L0_GPIO_MODE_OUTPUT));
            stm32l0_gpio_pin_write(RADIO_TCXO_VCC, 1);
        }
        else
        {
            stm32l0_gpio_pin_configure(RADIO_TCXO_VCC, (STM32L0_GPIO_PARK_NONE | STM32L0_GPIO_MODE_ANALOG));
        }
    }
}
```

evandavey, 2021-07-23:
But this is a separate issue to deep sleep (though does imply this library has a more detailed implementation so more likely to minimise current consumption).

evandavey, 2021-07-24:
I have had a break-through and am now getting down to 7uA!

I created a new SX1276 radio class which extends the original and overrides the sleep() and wakeup() methods.

I have only tested sleep but it looks like this so far:

```c++
 void sleep() final
{
    // call actual sleep method
    SX126X_LoRaRadio::sleep();

    // Set low power for pins

    GPIO_InitTypeDef GPIO_InitStructure = {0};

    // Disable ant switch (PA_15)
    GPIO_InitStructure.Mode = GPIO_MODE_ANALOG;
    GPIO_InitStructure.Pull = GPIO_NOPULL;
    GPIO_InitStructure.Pin = (GPIO_PIN_15);
    __GPIOA_CLK_ENABLE();
    HAL_GPIO_Init(GPIOA, &GPIO_InitStructure);
    __GPIOA_CLK_DISABLE();

    // Disable SPI pins (SCLK PB_13, MISO PB_14, MOSI PB_15)
    GPIO_InitStructure.Mode = GPIO_MODE_ANALOG;
    GPIO_InitStructure.Pull = GPIO_NOPULL;
    GPIO_InitStructure.Pin = (GPIO_PIN_13 | GPIO_PIN_14 | GPIO_PIN_15);
    __GPIOB_CLK_ENABLE();
    HAL_GPIO_Init(GPIOB, &GPIO_InitStructure);
    __GPIOB_CLK_DISABLE();
}
```
I have tried with CS (PB_12) with various pull up settings but this always results in ~1mA so assume this is turning the radio back on.  Don't know if this accounts for the missing 5uA.

Don't know if there is a better MBED way of doing this in PeripheralPins.c.  Also not sure if there are any global sleep / wakeup hooks that could be used instead as part of the sleep manager.

Will also need to work-out how to reverse in the wakeup() override.

evandavey, 2021-07-24:
HW deinit seems like it might be an outstanding mbed issue https://github.com/ARMmbed/mbed-os/issues/3106
@jeromecoutant  - any way to deinit / init SPI just for an STM32L0 target?

hallard, 2021-07-26:
> I have had a break-through and am now getting down to 7uA!
> 

Wow impressed, I think you're very close. It's the first time I see that changing SPI pins before sleep has an effect on sleep mode consumption (I mean with this module I remember the need on doing that with custom boards with RFM95) . I had same thing for CS Line. Did you tried to put it in HI-Z mode instead?

The grumpyoldpizza does not change these pin and it works,, correct?

By the way if there is another device on SPI bus it may affect it not but it's out of the scope of this issue of course.

evandavey, 2021-07-26:
I think GrumpyOldPizza does change these pins, and so does I-CUBE-LRWAN (see e.g. Drivers/BSP/SX1262DVK1CAS/sx1262dvk1cas.c).

```c++
static void SX1262DVK1CAS_RADIO_SPI_IoDeInit(void)
{
  GPIO_InitTypeDef GPIO_InitStruct;

  /* Peripheral clock disable */
  /* no need to call SPI1_CLK_DISABLE() because going in LowPower it gets disabled automatically */

  /* DeInitialize Peripheral GPIOs */
  /* Instead of using HAL_GPIO_DeInit() which set ANALOG mode
     it's preferred to set in OUTPUT_PP mode, with the pins set to 0 */

  GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
  GPIO_InitStruct.Pull = GPIO_PULLDOWN;
  GPIO_InitStruct.Pin = RADIO_SPI_MOSI_GPIO_PIN;
  HAL_GPIO_Init(RADIO_SPI_MOSI_GPIO_PORT, &GPIO_InitStruct);
  GPIO_InitStruct.Pin = RADIO_SPI_MISO_GPIO_PIN;
  HAL_GPIO_Init(RADIO_SPI_MISO_GPIO_PORT, &GPIO_InitStruct);
  GPIO_InitStruct.Pin = RADIO_SPI_SCK_GPIO_PIN;
  HAL_GPIO_Init(RADIO_SPI_SCK_GPIO_PORT, &GPIO_InitStruct);

  HAL_GPIO_WritePin(RADIO_SPI_MOSI_GPIO_PORT, RADIO_SPI_MOSI_GPIO_PIN, GPIO_PIN_RESET);
  HAL_GPIO_WritePin(RADIO_SPI_MISO_GPIO_PORT, RADIO_SPI_MISO_GPIO_PIN, GPIO_PIN_RESET);
  HAL_GPIO_WritePin(RADIO_SPI_SCK_GPIO_PORT, RADIO_SPI_SCK_GPIO_PIN, GPIO_PIN_RESET);
}

static void SX1262DVK1CAS_RADIO_SPI_IoInit(SPI_HandleTypeDef *spiHandle)
{
  GPIO_InitTypeDef GPIO_InitStruct;
  /* USER CODE BEGIN SPI1_MspInit 0 */

  /* USER CODE END SPI1_MspInit 0 */
  /* Enable Peripheral clock */
  RADIO_SPI_SCK_GPIO_CLK_ENABLE();
  RADIO_SPI_MOSI_GPIO_CLK_ENABLE();
  RADIO_SPI_MISO_GPIO_CLK_ENABLE();

  /**SPI1 GPIO Configuration
  PA5     ------> SPI1_SCK
  PA6     ------> SPI1_MISO
  PA7     ------> SPI1_MOSI
    */
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_PULLDOWN;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;

  GPIO_InitStruct.Alternate = RADIO_SPI_MOSI_GPIO_AF;
  GPIO_InitStruct.Pin = RADIO_SPI_MOSI_GPIO_PIN;
  HAL_GPIO_Init(RADIO_SPI_MOSI_GPIO_PORT, &GPIO_InitStruct);
  GPIO_InitStruct.Alternate = RADIO_SPI_MISO_GPIO_AF;
  GPIO_InitStruct.Pin = RADIO_SPI_MISO_GPIO_PIN;
  HAL_GPIO_Init(RADIO_SPI_MISO_GPIO_PORT, &GPIO_InitStruct);
  GPIO_InitStruct.Alternate = RADIO_SPI_SCK_GPIO_AF;
  GPIO_InitStruct.Pin = RADIO_SPI_SCK_GPIO_PIN;
  HAL_GPIO_Init(RADIO_SPI_SCK_GPIO_PORT, &GPIO_InitStruct);
}
```

I noticed that hal_deepsleep() is declared weak in TARGET_STM/sleep.c so have attempted to implement a custom deinit and init function (like above) before/after sleep but weirdly this behave as expected and I see very high current or no impact.  But this might be the most promising place to put it as it doesn't require extending the driver.

evandavey, 2021-07-29:
@hallard 

Can you please confirm your test measurement setup?  I am seeing an average of about 620uA on a SeeedStudio LoRa E5 Mini using your code from https://github.com/ARMmbed/mbed-os-example-lorawan/issues/227#issuecomment-885491038 and mbed-os-example-lorawan as the starter project.

I am using the target files from https://github.com/ARMmbed/stm32customtargets with the LORA_E5_MINI target.  

I am measuring power using a Nordic PPK2 powering the board at 3.3V via the 3.3V pin and GND.  Nothing else is connected.

I am compiling using GCC_ARM (GNU ARM 9-2019-q4-major) and with the Release profile.

hallard, 2021-07-29:
I confirm code mentioned (basic no full sample LoRawan not tested yet) consumption. Take care LoRa E5 mini have all usb stuff on board including usb serial chip connected to uart. depending on type and if this chip powered or not it could draw some current same as on board regulator where you apply 3.3v on it output

evandavey, 2021-07-29:
I had a look at the schematic and I think the USB circuitry is only powered when 5V is present.  The assert above has me thinking the radio isn't getting into sleep mode due to some error.  What mbed-os commit are you using?

evandavey, 2021-07-29:
I dead a dead-bug board.  I have broken out VCC,GND,SWCLK, SWDIO, NRST.  VCC and GND have a 1u and 100nF capactior on as short leads as possible.  Nothing else is broken out or connected.  Using the LORA E5 MINI as the target, I am still getting the same results as the actual mini.  I haven't connected the UART or a LED so I can't actually prove the code is running aside from that it programs and current matches.

@hallard - perhaps you could send me your compiled binary and I can test that?

hallard, 2021-07-29:
So even with only LoRa-E5 nothing else except caps (no regulator of course) you still have the issue? I can try back and give the compiled code but not today as I got so much other things to finish before.

evandavey, 2021-07-29:
@hallard  - no worries, whenever you can.  Yes, just the LoRa-E5 powered by the Nordic PPK2 @ 3.3V.

hallard, 2021-07-29:
Ok took some time here, used  latest master from 
https://github.com/ARMmbed/mbed-os
https://github.com/ARMmbed/stm32customtargets

then use this code (as release of course)
[image]

`mbed_app.json`
```json
{
    "requires": ["bare-metal", "events", "lora", "mbedtls", "stm32wl-lora-driver"],
    "target_overrides": {
        "*": {
            "target.printf_lib": "std"
        }
    }
}
```

`custom_targets.json`
```json
{
    "LORA_E5": {
        "inherits": [
            "MCU_STM32WLE5xC"
        ],
        "device_name": "STM32WLE5JCIx"
    },
    "LORA_E5_BREAKOUT": {
        "inherits": [
            "LORA_E5"
        ],
        "macros_add": [
            "LED1=PB_5", 
            "LED2=PB_10",
            "BUTTON1=PB_13"
        ]
    }
}
```

Connected as follow
[image]

Powering with 
[image]

and tadaaaaaa 1.5uA
[image]

Please note the 3mA peaks (mainly some OS wake) an if I put them into the selection I have average of 7.5uA
[image]

Hope this helps

evandavey, 2021-07-29:
@hallard  - thank you.  Very odd, our different results and almost identifcal setups.   I have installed MBED Studio and will try compiling with that tomorrow (very late here) and see if that changes anything.

hallard, 2021-07-29:
Here the compiled file if you want to give a try [mbed-os-deepsleep-test.bin.zip](https://github.com/ARMmbed/mbed-os-example-lorawan/files/6900955/mbed-os-deepsleep-test.bin.zip)

[39 more replies]